StefanSchippers/xschem #71: Question about window focus

rafmag, 2022-08-10:
I'm not sure if this is something you control, or if its a property of Linux and X11. But I noticed that if the mouse moves away from a window the window loses focus. This is extremely annoying when trying to type if you accidently move the mouse away from a text box. 

Do you know anything about this? Can this be fixed?

StefanSchippers, 2022-08-10:
this is a window manager task. the focus model. you can set 'focus follows
mouse', 'click to focus' and various additional quirks and options. you
should find these settings in your desktop control center.

rafmag, 2022-08-11:
hmm, except this behavior only happens within and between XSCHEM windows and nothing else. For example the schematic window and the property text window.

btw I don't have a desktop control center in this case. I'm running this on WSL2 using X11

rafmag, 2022-08-11:
Actually you know what, it ONLY happens when you move your mouse over the schematic area. That's when it pulls focus away from any other xschem window, like an object property window.

StefanSchippers, 2022-08-11:
thank you, will take a look...

StefanSchippers, 2022-08-11:
Can you please locate these 2 lines in xschem.tcl and remove the `focus $topwin` ?
If things work as you expect i will commit the change:
```
    # bind $topwin <Motion> "focus $topwin; xschem callback %W %T %x %y 0 0 0 %s"
    bind $topwin <Motion> "xschem callback %W %T %x %y 0 0 0 %s"
    # bind $topwin <Enter> "destroy .ctxmenu; focus $topwin; xschem callback %W %T %x %y 0 0 0 0"
    bind $topwin <Enter> "destroy .ctxmenu; xschem callback %W %T %x %y 0 0 0 0"
```

StefanSchippers, 2022-08-11:
Just a comment. if you select a component and press 'q' (or double click on it) and do some changes in the attributes, clicking in the main window another component will update the attributes of previous component and start editing the next. 
Clicking an empty area of the schematic will instead close the edit attribute window, updating the component. Pressing Esc key is equivalent to cancel button.

StefanSchippers, 2022-08-11:
The reason for the main window taking focus was to allow the user move/zoom/pan the schematic (dragging the mouse, using arrows, using mouse wheel) while editing a component attribute list, for example if trying to set transistor W equal to another transistor in the schematic, you double click the transistor to bring up the dialog box, then go in the schematic window and move /zoom until you see the other transistor so you can set the same W.
I have added xschemrc variable `autofocus_mainwindow` (default setting: enabled (1) ), if this variable is set moving the mouse to the main drawing area when a dialog box is open (typically while editing a component attribute list) will focus the main window, so user can move / zoom/pan the schematic using the usual commands, for example to look at a different part of the schematic while editing the component. If this variable is unset a click is needed to focus the main window. This will thus not allow to move / zoom, but allows to type into the dialog box if the mouse goes out of it.

rafmag, 2022-08-11:
Perfect! Good idea making it a configurable option in case others prefer it the way it was. I like it this new way better.

closed 2022-08-11 by rafmag