StefanSchippers/xschem #72: Issue plotting the result of a XYCE .step sweep internally.

rafmag, 2022-08-11:
I ran a minimum example .step sweep using xyce (see attached .raw file). when I plot this in BeSpice it looks like this:

[image]

But when I try to load the data into xschem. I get the following error: "read_rawfile(): no useful data found"

[STEP_TEST.zip](https://github.com/StefanSchippers/xschem/files/9310222/STEP_TEST.zip)

StefanSchippers, 2022-08-11:
I need to add this. Xyce has this plot header that xschem is unaware of:
`Plotname: Step Analysis: Step 1 of 6 params:  name = TEMP value = -40  DC transfer characteristic`
while the known (mostly ngspice) headers are:
```
Plotname: Transient Analysis
Plotname: DC transfer characteristic
Plotname: Operating Point
Plotname: AC Analysis
```
So Xyce inserts some more descriptive text before the `DC transfer characteristic` string.
Will fix this.

StefanSchippers, 2022-08-11:
I have pushed a fix for this, your sample file is read correctly now. 
Multiple datasets (multiple sweeps) is still handled in a bit primitive way by xschem, however if you double click the graph to bring up the graph dialog box you can enter a number in the Dataset text box and interactively see the relevant trace.

Thank you for sending me these tests with Xyce!. I have not yet started using Xyce extensively (although i have it installed).
[image]

rafmag, 2022-08-12:
> Thank you for sending me these tests with Xyce!. I have not yet started using Xyce extensively (although i have it installed).

No problem! My primary objective with all of this is to put together a set of tools that will help circuit designers more easily use Xyce.  So if you need something tested with Xyce or have any ideas you would like me to communicate to the Xyce team, let me know! You have been very helpful!

rafmag, 2022-08-12:
Working!

closed 2022-08-12 by rafmag