StefanSchippers/xschem #74: Better way for sending lots of commands to BeSpice Wave?

rafmag, 2022-08-14:
I can use a launcher to send a command to bespice like this:

`puts $bespice_server_getdata(sock) {set_curve_style "poweramp.raw" "v(outp)" "solid_line" "no symbol" 1 0 255 0}`

But what if I have a bunch of commands, say in a text file, one per line. Is there a way to send all of these commands? Have the program read the file, append the  `puts $bespice_server_getdata(sock) {` to every line and send it?

something like: `bespice_sendCommandFile(cmd.txt)`

It is a convenient way of saving a plot configuration with the necessary post-processing.

rafmag, 2022-08-14:
I take this back. Using the native tcl command `source script.tcl` lets me do what I want and gives me the flexibility to make more complex TCL scripts. Adding the `$bespice_server_getdata(sock) ` to BESPICE commands is minimal effort.

closed 2022-08-14 by rafmag

StefanSchippers, 2022-08-14:
You can use a couple of tcl lines to send a file named Bespice.cmd to the viewer:

set lines [split [read [set fd [open Bespice.cmd]]] \n]; close $fd
foreach line $lines { puts $bespice_server_getdata(sock) $line }

StefanSchippers, 2022-08-14:
note that in xschemrc you can preload a set of tcl scripts:
see following section and add a lappend line.
```
###########################################################################
#### TCL FILES TO LOAD AT STARTUP
###########################################################################
#### list of tcl files to preload.
# lappend tcl_files ${XSCHEM_SHAREDIR}/change_index.tcl
# lappend tcl_files ${XSCHEM_SHAREDIR}/ngspice_backannotate.tcl

```

rafmag, 2022-08-14:
> note that in xschemrc you can preload a set of tcl scripts that will be sourced after initialization is complete.

My intent is was to automate some of the post-processing of the waveforms after the simulation is complete. So I needed to be able to launch the script from the schematic. But I can do that with the `source` command.

StefanSchippers, 2022-08-14:
Yes i got your intent. Usually you create a tcl script with some useful procedure, this will be sourced at startup, then you can just execute the procedure from the schematic launcher.

rafmag, 2022-08-14:
Oh got it. So I could make my own TCL functions to process a BESPICE script file and then launch the script more easily from the schematic.

rafmag, 2022-08-14:
I have to start thinking like a programmer :)

StefanSchippers, 2022-08-14:
Yes. Also including the source command in the launcher is overkill. It will be sourced over and over again, taking more time to parse the script to bytecode.