StefanSchippers/xschem #75: Question about the derivative function.

rafmag, 2022-08-16:
I'm working with the BeSpice Wave author, trying to get the derivative function implemented in the post-processing calculator. I liked the way it works in Xschem. Are you using any special algorithm for the numerical differentiation? I know I could look through the code and figure it out, but I thought I'd ask.

StefanSchippers, 2022-08-16:
The assumption i made for derivative calculatrions is that signals are properly sampled in raw files, in particular on sharp edges timesteps are reduced / more dense. 
So my derivative calculation is a simple scheme:

`df(x) / dx @ x=x(n)  = f(x(n)) - f(x(n-1)) / (x(n) - x(n-1))`
with these special (arbitrary) assumptions:
`df(x(0) / dx = 0`
Also if two successive points in raw file have the same sweep value (it has happened, that is x(n) = x(n-1) )
confirm the previous calculated derivative.

rafmag, 2022-08-16:
Very interesting, I've only ever used the derivative function when estimating the output resistance of a circuit from an IV curve, or for gain like you just showed. I'm not sure the precision of a 3 point derivative is needed. I wonder how other tools do this. Thanks for the info!

closed 2022-08-16 by rafmag