StefanSchippers/xschem #76: Conditional Format attribute on a symbol

rafmag, 2022-08-17:
Is there a way to have a conditional format attribute on a symbol?

For example, I have a symbol for a sub circuit that normally has 3 pins, but depending on it's use case I might want an extra pin to show up in the netlist and be connected to a variable using the "extra" option. But I don't want to have to change out the symbol, just base it on the value of a TCL set variable. like an `if-then` for a format attribute. 

Example:
TCL:  SET MODEL_TYPE 1

if  MODEL_TYPE==1 then Symbol:
 `format="#@designator\\#@name @@pin1 @@pin2 @@pin3 @model"`
`template="name=m1 designator=X model=modelname pin4=pvar"`
`extra="pin4"`

if  MODEL_TYPE==2 then Symbol:
 `format="#@designator\\#@name @@pin1 @@pin2 @@pin3 @@pin4 @model"`
`template="name=m1 designator=X model=modelname pin4=pvar"`
`extra="pin4"`

The only change is in the format attribute. Assuming pvar will be a valid global node when in use.

I'm guessing there is some sort of TCL command I can put inside `format`, but for the life of me, I can't figure it out.

Is something like this possible? I realize it's a weird request, but the alternative is a mayor change to Xyce that might be a hard sell.

rafmag, 2022-08-17:
I think I got it, unless there is an issue I'm overlooking:

```
format="tcleval([expr { bool($PIN4_ENABLE) ? {#@designator\\#@name @@P1 @@P2 @@P3 @P4 @model } : {#@designator\\#@name @@P1 @@P2 @@P3 @model } }])
"
template="name=m1
designator=X
P4=p4net
model=modelname"
extra="p4"
```
Thoughts?

StefanSchippers, 2022-08-17:
That is correct. 
i have done this to make sure that if symbol is a subcircuit also its expansion is consistent:

```
type=subcircuit
format="tcleval([expr {$PIN4_ENABLE ? { @name @@A @@B @@Z @PIN4 @symname } : { @name @@A @@B @@Z @symname }}])"
template="tcleval(name=X1 [expr {$PIN4_ENABLE ? {PIN4=GND} : {}}])"
extra="tcleval([expr {$PIN4_ENABLE ? {PIN4} : {}}])"

```
By trying your advanced usages i have updated xschem, in some cases the attribute parsing in netlisting code was not passing the strings to tcl (if tcleval() was present) so above example was not fully working. I have now make the code consistent. tcleval() is always honored by passing the internal string to tcl.

In some next commits i will do also some better error checking. In some cases i had a xschem crash while using non valid tcl code in attributes (like missing braces etc). will fix that.

rafmag, 2022-08-17:
Yes, the crashes happened to me too. I wasn't sure if it was some issue with my WSL+XMING setup or if it was an XSCHEM issue.

Btw, I really like that you have TCL scripts embedded all the way into the symbol attributes. It makes everything very flexible/customizable! It lets me do some crazy stuff that's not possible with other schematic capture tools.

closed 2022-08-17 by rafmag

StefanSchippers, 2022-08-17:
Thank you!
yes the crashes need to be fixed.. the tcleval() does a catch to trap
errors and returns "?" if script has errors. But there is s'thing more to
be checked.

Another feature that is still missing is to instantiate a symbol as
opamp.tcl(par1, par2, ...)
or opamp.py()
or opamp.awk()

In these cases xschem should execute the referenced tcl, python, awk script
with the optional parameters. the script produces on standard output the
xschem symbol. A kind of symbol generator.

StefanSchippers, 2022-08-17:
Adding a tcl parsing is a great boost on flexibility.
However passing everything to tcl just because there might be a  tcl
expression to be evaluated (in 98% of cases there is not) is super
overkill. So i just do a string check for tcleval() which is super fast in
C and hand over to the tcl interp only if that tcleval() wrapper is found.