StefanSchippers/xschem #77: TCL command for LVS SPICE Export

rafmag, 2022-08-19:
I'm trying to figure out a TCL command that would allow me to export the spice file for a subcircuit whose instance I specify in LVS format (no spice designators for devices and in subcircuit format)

For example, assume x1 is an instance of an opamp in a test bench. Inside the opamp the transistors are subcircuits xm1, xm2...

Example: tclcommand= ExportInstanceLVS(x1)

Result:
.subcircuit OpAmpSubcircuitModel pin1 pin 2 pin 3
   m1 net1 net 2 net3 mSubcircuitModel
   .
   .
   .
.ends

Any Ideas?

StefanSchippers, 2022-08-20:
As far as i understand the following bunch of tcl commands will do the job:

```
set spiceprefix 0
set top_subckt 1
xschem select instance x1
xschem descend
xschem netlist
xschem go_back
```

Thank you for asking. By looking at your advanced user questions i found a regression in the code. The spice netlist code was incorrectly resetting the spiceprefix variable to 1, overriding user choice. Fixed just now.

StefanSchippers, 2022-08-20:
Another possibility if to handover the netlist creation  to another xschem process. You can launch the following from the command line:
`xschem -n -s -x -r -q --tcl 'set spiceprefix 0; set top_subckt 1' sky130_tests/bandgap_opamp.sch`
-n: do a netlist
-s: spice format
-x: do not startup Xcode (windows etc)
-r: do not load tclreadline command line editing features
-q: quit after doing requested command.
--tcl: execute tcl script before doing requested command

StefanSchippers, 2022-08-20:
the above can be launched directly from the xschem prompt:
`execute 0 xschem -n -s -x -r -q --tcl {set spiceprefix 0; set top_subckt 1} sky130_tests/bandgap_opamp.sch`
the 0 after execute tells xschem not to report in a dialog box the exit status of the command.

a more general command that gets the symbol flname from the instance name (x1) is:
`execute 0 xschem -n -s -x -r -q --tcl {set spiceprefix 0; set top_subckt 1} [file rootname [xschem getprop instance x1 cell::name]].sch`

rafmag, 2022-08-20:
I have an issue with spiceprefix now related to  #76 I'm using the following format for the name:
`#@spiceprefix\\#@name`
This works well when spiceprefix is enabled, but when its not the netlist now has `##m1` instead of just `m1`

StefanSchippers, 2022-08-20:
The spiceprefix does not need to be wrapped inside \#...\#, it will be correctly added when expanding array devices.
This attribute is a bit special since it can be enabled/disabled at global level, so it is handled a bit differently.
Just use `@spiceprefix@name`

rafmag, 2022-08-20:
spiceprefix is not expanding to array devices after I removed the #...#

StefanSchippers, 2022-08-20:
A good addition would be the netlister to drop the hashes (##) if the @attribute inside #@attribute# is undefined or empty. will look into it.

rafmag, 2022-08-20:
> A good addition would be the netlister to drop the hashes (##) if the @Attribute inside #@Attribute# is undefined or empty. will look into it.

You might have to do this, since 

> spiceprefix is not expanding to array devices after I removed the #...#

StefanSchippers, 2022-08-20:
This is strange.
I have tried the following device:
[image]

and the spiceprefix is added to all array instances:
```
XM12[4] net10[4] net7[4] net9[4] net8[4] sky130_fd_pr__pfet_01v8_lvt L=2 W=8 nf=1 ad='W * 0.29' as='W * 0.29'
+ pd='2*(W + 0.29)' ps='2*(W + 0.29)' nrd=0 nrs=0 sa=0 sb=0 sd=0 mult=1 m=1
XM12[3] net10[3] net7[3] net9[3] net8[3] sky130_fd_pr__pfet_01v8_lvt L=2 W=8 nf=1 ad='W * 0.29' as='W * 0.29'
+ pd='2*(W + 0.29)' ps='2*(W + 0.29)' nrd=0 nrs=0 sa=0 sb=0 sd=0 mult=1 m=1
XM12[2] net10[2] net7[2] net9[2] net8[2] sky130_fd_pr__pfet_01v8_lvt L=2 W=8 nf=1 ad='W * 0.29' as='W * 0.29'
+ pd='2*(W + 0.29)' ps='2*(W + 0.29)' nrd=0 nrs=0 sa=0 sb=0 sd=0 mult=1 m=1
XM12[1] net10[1] net7[1] net9[1] net8[1] sky130_fd_pr__pfet_01v8_lvt L=2 W=8 nf=1 ad='W * 0.29' as='W * 0.29'
+ pd='2*(W + 0.29)' ps='2*(W + 0.29)' nrd=0 nrs=0 sa=0 sb=0 sd=0 mult=1 m=1
XM12[0] net10[0] net7[0] net9[0] net8[0] sky130_fd_pr__pfet_01v8_lvt L=2 W=8 nf=1 ad='W * 0.29' as='W * 0.29'
+ pd='2*(W + 0.29)' ps='2*(W + 0.29)' nrd=0 nrs=0 sa=0 sb=0 sd=0 mult=1 m=1
```

rafmag, 2022-08-20:
I'm using it inside a tcleval as we discussed in #76, maybe that has something to do with it?

StefanSchippers, 2022-08-20:
Ok will check. Thank you

StefanSchippers, 2022-08-20:
Please update again xschem, i found another glitch in the netlisting code (but not related to spiceprefix addition in array devices).

If i have a symbol with following attributes:

```
type=subcircuit
format="tcleval(
  [expr {$PIN4_ENABLE ? {@spiceprefix@name @@A @@B @@Z @PIN4 @symname\\} :
                        {@spiceprefix@name @@A @@B @@Z @symname\\}}]
)"
template="tcleval(name=I1[expr {$PIN4_ENABLE ? { PIN4=GND} : {}}] spiceprefix=X)"
extra="tcleval([expr {$PIN4_ENABLE ? {PIN4} : {}}])"
```
and instantiate the symbol as an array:
[image]
netlist seems correct (in this case i did a ```set PIN_ENABLE 1```):
```
XI1[4] AA BB ZZ GND cond_symbol
XI1[3] AA BB ZZ GND cond_symbol
XI1[2] AA BB ZZ GND cond_symbol
XI1[1] AA BB ZZ GND cond_symbol
XI1[0] AA BB ZZ GND cond_symbol
```
so it seems ok.

rafmag, 2022-08-20:
I did an almost copy paste, the only difference is that my symbol type is not subcircuit and it didn't propagate to the array devices.

StefanSchippers, 2022-08-20:
Have you tried to enable it in Simulation -> Use 'spiceprefix' attribute?

rafmag, 2022-08-20:
Yes

rafmag, 2022-08-20:
the first device in the array has the spiceprefix but the rest don't.

StefanSchippers, 2022-08-20:
Even if i set my symbol type to something different than 'subcircuit' (so that sub schematic is not expanded) it does fine:
So if that does not work for you can you please send me the offending symbol?

StefanSchippers, 2022-08-20:
The only way to get a behavior like yours is to use a different prefix attribute for that, like `spicepref` or `spice_prefix`. In this case it is treated as an ordinary attribute and not prefixed to all aray instances unless wrapped inside hashes (#@spice_prefix#)

rafmag, 2022-08-20:
Ok it's messing with me... I tried to create a new file for you. And it behaved normally. So removed every instance and re-inserted them and now it all works!

One last thing:

I'll likely use this approach:
`execute 0 xschem -n -s -x -r -q --tcl {set spiceprefix 0; set top_subckt 1} [file rootname [xschem getprop instance x1 cell::name]].sch`

What variable do I set to enable LVS netlist?

rafmag, 2022-08-20:
Also in the command, is there a way to choose where to save the netlist?

StefanSchippers, 2022-08-20:
the `set top_subckt 1` is the underying global variable that is set when enabling LVS netlist. for the output directory add `-o path` to command line. To rename the netlist add `-N xxx.spice`. Example, produce xxx.spice in /tmp:

`execute 0 xschem -n -s -x -r -q -o /tmp -N xxx.spice --tcl {set spiceprefix 0; set top_subckt 1} [file rootname [xschem getprop instance x1 cell::name]].sch`

rafmag, 2022-08-20:
Perfect, this is exactly what I needed! Thanks for the help as always.

closed 2022-08-20 by rafmag

StefanSchippers, 2022-08-20:
A bit of background:

the spiceprefix attribute was created expecially for spice netlists where a 'X' prefix is required for subcircuits. Since many users prefer I1, I2, ... or U1, U2, ... for their reference designators (name attribute in xschem) this allows to have a valid spice netlist.

The `#@attribute\\#` (or without a variable, like: `#AAAA#` ) mechanism is a generic way to tell the netlister code "add this prefix to every array element of the following field. This can be used to add a prefix to expanded array nodes as well, like in:
`format="#@spicepref\\#@name......   #X@spicepref\\X#@name ..."`
If name=I4[4:0] and spicepref=Z  each array line in spice netlist will have: 

```
ZI4[4] XZXI4[4] A B Z GND cond_symbol3
ZI4[3] XZXI4[3] A B Z GND cond_symbol3
ZI4[2] XZXI4[2] A B Z GND cond_symbol3
ZI4[1] XZXI4[1] A B Z GND cond_symbol3
ZI4[0] XZXI4[0] A B Z GND cond_symbol3
```
so the XZXI4[...] nodes are unique in the design. This allows to create multi line expansions for example create a nand2 gate by printing 4 transistor lines in the netlist. A nand gate has an internal node that needs to be unique. 

See the '`sky130_tests/test_nand2_noschematic.sch`' example added to the [xschem_sky130](https://github.com/StefanSchippers/xschem_sky130)   example circuits

StefanSchippers, 2022-08-20:
Fixed also the removal of '##' in final netlist if the internal attribute of  `#@attribute\\#` is undefined or empty

rafmag, 2022-08-20:
> Fixed also the removal of '##' in final netlist if the internal `attribute` of `#@attribute\\#@name` is undefined or empty

Thanks, I'm sure this will be useful as well.

StefanSchippers, 2022-08-20:
Thank you for testing all these advanced user features. It helps me a lot to improve the tool.

urish, 2025-02-08:
Note for future wanderers finding this issue: `top_subckt` is now called `lvs_netlist`