StefanSchippers/xschem #78: Unique Instance names...

rafmag, 2022-08-21:
I noticed that sometimes when you copy instances the instance name does not increment correctly and you end up with duplicate instance names within a sub circuit.

Having non-unique instance names can sometimes cause issues. If this is not easy to fix, it would be nice if XSCHEM can warn you about duplicate instance names, just like it warns you about disconnected pins.

StefanSchippers, 2022-08-21:
Xschem does check to avoid duplicate instance names, one case where collisions are possible is mixed case name attributes.

A x1 instance may coexist with a X1 instance. For Verilog (which is case insensitive) this is valid. For spice this is a name collision.
There is a check for duplicate instance names in xschem:
Highlight -> Highlight duplicate instance names (# key)
However it will flag exact name collisions. May be i can set this to be case insensitive.

If after a copy operation you end up with exact (same case) instance names let me know how you did it, this is a bug.

StefanSchippers, 2022-08-21:
Another possible issue is a device called X1[3:0] and one called X1[2]: these 2 names are different, however after expansion there will be a collision. This is not detected. One possible solution is to enforce unique name radix (the part before the '['). However this will not allow you to place X1[3] and X1[2] separately. A better fix is to do expansion (expand X1[3:0] into X1[3], X1[2], X1[1], X1[0]) and check individual 'bits'. This will take more time.

StefanSchippers, 2022-08-21:
A little improvement. 
Name collision check in xschem is now case insensitive.
if 'x1 and 'X2' are present in the design a new subcircuit will be placed with name X3 (if template string uses name =X1) or x3 (if template string uses name=x1)
Duplicate name check is now case insensitive. instances x1 and X1 will be flagged as duplicates.

Consider that there is an option in xschem to allow duplicate instance names (Symbol->allow duplicate instance names(refdes)). This was a request from PCB designers. For simulation purposes this option makes no sense.

rafmag, 2022-08-21:
I think this is good for now, I'll be on the look out for duplicates when copying instances and let you know.

closed 2022-08-21 by rafmag