StefanSchippers/xschem #79: TCL Script for traversing hierarchy...

rafmag, 2022-08-21:
How could I write a loop that traverses the hierarchy and prints out a list of instances with their relative names? For example: x1/x2/R4

I know I can loop through all instances like this:
```
set lastinst [xschem get instances]
for { set i 0 } { $i < $lastinst } {incr i } {
  puts [xschem getprop instance $i name]
}
```
But that just gives me the name of the instance with no relative path. example: R4 instead of x1/x2/R4

rafmag, 2022-08-21:
I just realized, that loop only gets me the top level instances. Would I need to do some sort of manual recursion here?

StefanSchippers, 2022-08-21:
Try to play with this: create this tcl file, source it into xschem, then give the command `traversal xxxx` where `xxxx` is a file to get the hierarchy written into.
```
proc traversal {file} {
  if { $file eq {} || [file exists $file] } {
    puts stderr "empty or existing file..."
    return
  }
  xschem unselect_all
  xschem set no_draw 1 ;# disable screen update
  xschem set no_undo 1 ;# disable undo 
  set fd [open $file "w"]
  hier_traversal $fd 0
  xschem set no_draw 0
  xschem set no_undo 0
  close $fd
}

# return "$n * $indent" spaces
proc spaces {n} {
  set indent 4
  set n [expr {$n * $indent}]
  return [format %${n}s {}]
}

proc hier_traversal {fd {level 0}} {
  set schpath [xschem get sch_path]
  set instances  [xschem get instances]
  for {set i 0} { $i < $instances} { incr i} {
    set instname [xschem getprop instance $i name]
    set symbol [xschem getprop instance $i cell::name]
    set type [xschem getprop symbol $symbol type]
    puts $fd "[spaces $level]$schpath$instname symbol: $symbol, type: $type"
    if {$type eq {subcircuit}} {
      xschem select instance $i
      set current_level [xschem get currsch]
      xschem descend 1
      # ensure previous descend was successful
      if {[xschem get currsch] == $current_level + 1} {
        incr level
        hier_traversal $fd $level
        xschem go_back
        incr level -1
      }
    }
  }
}

```

rafmag, 2022-08-21:
I tried something like this:
```

proc printInst {path} {
    set lastinst [xschem get instances]
    for { set i 0 } { $i < $lastinst } {incr i } {
      set name [xschem getprop instance $i name]
      set type [xschem getprop instance $i cell::type]
      if { $type eq {subcircuit} } {
        xschem select instance $i
        xschem descend $i
        printInst $path/$name
        xschem go_back
      }
      puts $path/$name
    }
}
```

It seemed to work but only if I had `xschem set no_draw 0` for some reason.

StefanSchippers, 2022-08-21:
setting `xschem set no_draw 1` is only done for performance reasons and to avoid screen flickering while xschem traverses the circuit. 
However if for some reason you don't set it again to 0 (`xschem set no_draw 0`) at the end you get a frozen screen.
xschem is working normally but you can't see any screen update.
If in doubt don't set this .

rafmag, 2022-08-21:
Looking at your code, I realized I misinterpreted `xschem descend` fixed now. 

This works now. start at the desired root instance:

```
proc printInst {path} {
    set lastinst [xschem get instances]
    for { set i 0 } { $i < $lastinst } {incr i } {
      set name [xschem getprop instance $i name]
      set type [xschem getprop instance $i cell::type]
      if { $type eq {subcircuit} } {
        xschem select instance $i
        xschem descend 1
        printInst $path/$name
        xschem go_back
      }
      puts $path/$name
    }
} 
```

closed 2022-08-21 by rafmag

StefanSchippers, 2022-08-21:
Yes the xschem descend commandaccepts an instance iteration to descend into, if you have vector instances like X1[15:0]. In this case the `xschem descend 1` command  only descends into the leftmost instance X1[15]. 

I found another bug in the code thanks o your advanced usage!

my intention was that `xschem descend -1` should go into the rightmost instance (X1[0]  in above example). It didn't wor, now it is fixed.

StefanSchippers, 2022-08-21:
In case of vector instances a true full traversal should loop thru all vector elements. In case of `X1[15:0]` a loop of `xschem descend $n` where n goes from 1 to 16 is required.

StefanSchippers, 2022-08-21:
For the above task there is a useful `xschem expandlabel` command:

if `instname` contains `X1[3:0]`
the command
`xschem expandlabel $instname `
returns:
`X1[3],X1[2],X1[1],X1[0] 4`
the last field is the number of elements.

rafmag, 2022-08-21:
Interesting. what does expandlabel return if it's given a normal instance name `xschem expandlabel m1`?

StefanSchippers, 2022-08-21:
m1 1

StefanSchippers, 2022-08-21:
last field is always the number of instances.

StefanSchippers, 2022-08-21:
you can extract this number with:
`set ninst [lindex [split [xschem expandlabel $instname] { }] 1]`
and do the descend loop:
```
for {set n 1} {$n <= $ninst} { incr n{} {
  xschem descend $n
  ....
}
```

StefanSchippers, 2022-08-21:
You can then optimize the script. If you have a X1[1023:0] subcircuit descending 1024 times into it is probably extremely slow and inefficient. 
It is much better to descend one single time, get a list of the inner components and print `ninst` times with the path built up appropriately.