StefanSchippers/xschem #80: Custom Toolbar Button...

rafmag, 2022-08-27:
How can I add a custom toolbar button with my own TCL command?
Where is the icon and callback defined?

StefanSchippers, 2022-08-27:
Images are defined in file resources.tcl, encoded as base64 ascii data. There are some notes in the file on how to create these.
Following piece of code (put it into add_toolbar_button.tcl) adds a custom toolbar button:
```
# This example shows how to add a custom toolbar to xschem
# Stefan Schippers 20220827

# 24x24 image encoded as gif (so it will be loaded by 20 years old tcl8.4)
# transparent background, encoded as base64
set MyButtonData {
  R0lGODlhGAAYAPcAACIiIiMjIyUlJSYmJicnJykpKSoqKisrKywsLDAwMDExMTIyMjMzMzQ0NDU1
  NTY2Njc3Nzg4ODk5OTo6Oj09PT4+Pj8/P0FBQUJCQkNDQ0REREVFRUZGRklJSUxMTE1NTU5OTlFR
  UVJSUlNTU1lZWVpaWltbW1xcXF1dXV5eXl9fX2FhYWNjY2RkZGZmZmdnZ2hoaGlpaWtra2xsbG1t
  bXFxcXJycnR0dHV1dXl5eXt7e3x8fICAgIODg4eHh4uLi4yMjI2NjY6Ojo+Pj5GRkZKSkpSUlJaW
  lpeXl5mZmZqampubm5ycnJ2dnZ6enp+fn6GhoaKioqOjo6SkpKampqenp6ioqKmpqa6urrCwsLGx
  sbOzs7S0tLW1tba2tre3t7m5ubu7u76+vsDAwMHBwcLCwsbGxsfHx8jIyMnJycrKysvLy83Nzc/P
  z9DQ0NLS0tPT09XV1dbW1tfX19jY2NnZ2dra2tvb29/f3+Hh4eLi4uXl5efn5+jo6Onp6erq6uvr
  6+zs7O/v7/Dw8PX19fb29vf39/j4+Pn5+fr6+vv7+/z8/P39/f7+/gAAAAAAAAAAAAAAAAAAAAAA
  AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
  AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
  AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
  AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
  AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
  AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACH5BAEAAI4ALAAAAAAYABgA
  AAj+AB0JHDhwUR85ew4RXMhQoBgaCABIFJDiisKGBPGUkAhAAAQDHDGUwShQjQIACH68uahniYSO
  UTDaiViiT8NCOyRiYXgIAwAXZlKMoDJQjdArRwAYwLOwCQANdkBKzOIIj1QAXGgAkEFwEQQAYZyy
  CAKAhaMlP39s3QPSzsA2ADAsogIAhAwANBxFqXu3hiMcAI4MJAJghyNBEgR0VHP4Zcc2jriUHZiT
  qKM+QXi8Gdjnxw45Au08HaiVC0mGfQAooAzA8umBojUMBsDjNcEwAFIUHe2IRo2LBJuMsJkjcMGT
  ZRadlDGoIGEAcgRFBD0waYdCaiIiwHGERwWJTf5TTiZY6HuNRXJAcJQIIYzejtQJvgHpYtAiNUFy
  BOFyaBHZ1hiVEZECR+gxkCBXaNCRYCSlx5ECGnwlkQSmvbZIFiwoxhEISxRiG3l2mBHdhyQ2FBAA
  Ow==
}

# Add a "MyButton" button to toolbar_list
lappend toolbar_list MyButton

# Create an image object. Name should be img<name of button>
image create photo imgMyButton -data $MyButtonData

# Create the toolbar button
# constructor     name           tcl command       tooltip    topwindow
toolbar_create   MyButton      { puts SMILE! }     "SMILE!"     {}
#destroy and rebuild the toolbar
toolbar_hide
toolbar_show
```
in xschem source this file. You will see the new button (the smiling face):
[image]

StefanSchippers, 2022-08-27:
`add_custom_button.tcl` is now in the xschem repository as well as `add_custom_menu.tcl`.
All these customizations do not require any rebuild and can be done runtime, one more reason why i like tcl-tk.

rafmag, 2022-08-28:
Thanks for the info. I'll give it a try and report back.

StefanSchippers, 2022-08-28:
If you don't care about compatibility with older tcl-tk  libraries you can create a button image directly from a png file:
`image create photo imgMyButton -file MyButton.png`
I have added a forced 24 pixel height of the toolbar, so if your image is bigger it will not make the whole toolbar bigger, the image will be cropped.

StefanSchippers, 2022-09-29:
closed as no further feedback.

closed 2022-09-29 by StefanSchippers