StefanSchippers/xschem #81: Unable to use ./configure command

satvikgoel950, 2022-09-02:
after cloning the repo when i use ./configure it says
no directories found 
./conifgure 4 not found
I have installed all the libraries in the manual but still i face same error
what could be possible wrong please help

StefanSchippers, 2022-09-03:
Try to delete the cloned repository and repeat:
The steps are: 
```
git clone https://github.com/StefanSchippers/xschem xschem_git
cd xschem_git
./configure
make
sudo make install

```
Don't do any following steps if previous fails.

If this fails, please send me the file `xschem_git/scconfig/config.log`

satvikgoel950, 2022-09-03:
Thanks for replying, I have tried cloning again but no luck. Also in the folder told by you there is no config.log file. Attaching a screenshot of same.
[image]

StefanSchippers, 2022-09-03:
Please send me a listing of the xschem_git directory and a screenshot of the result of the ./configure command.

satvikgoel950, 2022-09-03:
Here they are.
[image]
[image]

StefanSchippers, 2022-09-03:
Ok, this is simple. You don't have a C compiler installed on the system. Install `gcc` package and also `make`, in case you don't have it already.

satvikgoel950, 2022-09-03:
But they are already installed. heres a screenshot
[image]

StefanSchippers, 2022-09-03:
Ok, your system does not have a cc symlink to gcc, so try this:

./configure --CC=gcc

satvikgoel950, 2022-09-03:
same error
[image]

StefanSchippers, 2022-09-03:
Oh yes, you need to have CC defined even before building configure itself.
Do this:
```
export CC=gcc
./configure
```

satvikgoel950, 2022-09-03:
same error again
[image]

StefanSchippers, 2022-09-03:
ok do these commands.
echo $PATH
which gcc
which cc
and send the results as text, don't do screenshots
also do: 
 ./configure >log 2>&1 
 and send me the log file (no screenshots)
 
 Thank you very much for reporting

satvikgoel950, 2022-09-03:
satvik@satvik:~/xschem_git$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
satvik@satvik:~/xschem_git$ 
satvik@satvik:~/xschem_git$ which gcc
satvik@satvik:~/xschem_git$ which cc
satvik@satvik:~/xschem_git$

StefanSchippers, 2022-09-03:
send the log file of `./configure >log 2>&1`

satvikgoel950, 2022-09-03:
these are the contents of log2 file
gcc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default   -c -o hooks.o hooks.c

file in not supported to upload

StefanSchippers, 2022-09-03:
there should not be a log2 file, there is a space after log, so please try again the above command:
./configure >log 2>&1

You will have a 'log' file.

satvikgoel950, 2022-09-03:
gcc -DGENCALL -DRUNTIME -g  -DPLUGIN_SCRIPTS   -Isrc//tmpasm  -DPLUGIN_PARSGEN    -DPLUGIN_GUI     -Isrc//default   -c -o hooks.o hooks.c
make: gcc: No such file or directory
make: *** [<builtin>: hooks.o] Error 127
./configure: 4: ./configure: not found

satvikgoel950, 2022-09-03:
yes these are the content unable to upload the log file

StefanSchippers, 2022-09-03:
ok thanks for now

StefanSchippers, 2022-09-03:
ok i am asking to more ubuntu users, 
can you please paste here (as text) the output of following command? 
Please close the terminal and open a new one , xince the export CC we did before must go away.
`update-alternatives --display cc`

satvikgoel950, 2022-09-03:
here,
satvik@satvik:~$ update-alternatives --display cc
update-alternatives: warning: alternative /usr/bin/gcc (part of link group cc) doesn't exist; removing from list of alternatives
cc - auto mode
  link best version not available
  link currently points to /usr/bin/gcc
  link cc is /usr/bin/cc
  slave cc.1.gz is /usr/share/man/man1/cc.1.gz
satvik@satvik:~$

StefanSchippers, 2022-09-03:
can you also do (copy and paste below text in a terminal):
dpkg -l |grep gcc

satvikgoel950, 2022-09-03:
atvik@satvik:~$ dpkg -l |grep gcc
ii  gcc                                        4:11.2.0-1ubuntu1                       amd64        GNU C compiler
ii  gcc-11                                     11.2.0-19ubuntu1                        amd64        GNU C compiler
ii  gcc-11-base:amd64                          11.2.0-19ubuntu1                        amd64        GCC, the GNU Compiler Collection (base package)
ii  gcc-12-base:amd64                          12-20220319-1ubuntu1                    amd64        GCC, the GNU Compiler Collection (base package)
ii  libgcc-11-dev:amd64                        11.2.0-19ubuntu1                        amd64        GCC support library (development files)
ii  libgcc-s1:amd64                            12-20220319-1ubuntu1                    amd64        GCC support library
ii  libuno-cppuhelpergcc3-3                    1:7.3.5-0ubuntu0.22.04.1                amd64        LibreOffice UNO runtime environment -- CPPU helper library
ii  libuno-purpenvhelpergcc3-3                 1:7.3.5-0ubuntu0.22.04.1                amd64        LibreOffice UNO runtime environment -- "purpose environment" helper
ii  libuno-salhelpergcc3-3                     1:7.3.5-0ubuntu0.22.04.1                amd64        LibreOffice UNO runtime environment -- SAL helpers for C++ library
satvik@satvik:~$

StefanSchippers, 2022-09-03:
Thank you for your patience
So it appears your system *thinks* gcc is installed and it was installed but /usr/bin/gcc is not there, for some reason installation is broken. Try to force-reinstall gcc and make:
`sudo apt-get install --reinstall gcc make`

satvikgoel950, 2022-09-03:
Thank you so much sir it worked.

StefanSchippers, 2022-09-03:
for some reason gcc was not installed correctly during ubuntu installation. You see how many weird problems may happen.

StefanSchippers, 2022-09-03:
Ok if xschem was built with no problems please close this issue. If you ran in other related problems you can open a new issue of course.

satvikgoel950, 2022-09-04:
yes. Thanks for the support.

closed 2022-09-04 by satvikgoel950