StefanSchippers/xschem #82: Parametrized subcircuit netlist issue

HarrySacha, 2022-09-07:
Hello Stefan,

I am facing the following issue:

I created a device StabProbeSE (schematic and symbol)

The symbol attributes in the symbol editor are:

type="subcircuit"
format="@name @pinlist @symname params: x=@x"
template="name=x1 x=1"

In the main schematic (where StabProbeSE is instanced as x1) x is assigned to {stb}, stb being a simulation parameter of a parametric analysis. The attributes of the subcircuit x1 in the schematic editor are:

name=x1 x={stb}

The issue is the following:

I would expect to see the subcircuit instance in the netlist written as follows:

x1 VGP2_2 VGP2_1 StabProbeSE params: x={stb} 

Instead I get:

x1 VGP2_2 VGP2_1 StabProbeSE params: x='stb'

For some reason the curly braces are replaced by single quotes. I tried a few workarounds but I failed.

Are you aware of this issue?

Thank you for attention. Best regards.

Serge Ramet

StefanSchippers, 2022-09-07:
curly braces are replaced by single quotes in the netlist post processing.
I believe ngspice recognizes both, and unless you give expressions like
x1='stb*2' you don't need quotes or curly braces at all. what simulator are
you using? the {} -> ' ' replacement was done many years ago because some
simulator did not accept braces. I can reconsider this today.

HarrySacha, 2022-09-09:
I am using xyce simulator. Thank you for your explanations. As a new user of xschem I was checking how the parameters were passed from sub-circuit  symbol attributes to the netlist, hence my query. My simulation works well using the xschem generated netlist so I close the issue.
Thanks and regards.

Serge Ramet

closed 2022-09-09 by HarrySacha