StefanSchippers/xschem #83: Problems Compiling and Running

kwmartin, 2022-09-21:
possibly be solved by adding lines like below to under install: in Makefile in src directory
chmod 755 $(find . -type d)
chmod 644 $(find . -type f)
or replace '.' with the appropriate directories. I done after install, needs sudo.
2) Can't run xschem as USER_CONF_DIR is not defined in Tcl_AppInit() at xinit.c:1928
I tried defining USER_CONF_DIR as an environment variable; didn't help.
I tried:
./configure --debug --user-conf-dir=~/.xschem when doing the make and I can see the definition in Makefile.conf:
BINDIR=$(DESTDIR)$(install_root)$(PREFIX)/bin
SHAREDIR=$(DESTDIR)$(install_root)$(PREFIX)/share
DOCDIR=$(DESTDIR)$(install_root)$(PREFIX)/share/doc
XSHAREDIR=$(SHAREDIR)/xschem
XDOCDIR=$(DOCDIR)/xschem
MANDIR=$(SHAREDIR)/man
USER_CONF_DIR=~/.xschem
SCCBOX=$(ROOT)/scconfig/sccbox
and make went fine, but still no good.
I also defined it in /usr/local/share/xschem/xschemrc using
set USER_CONF_DIR ~/.xschem
but still no go.

Can you give advice on how to get around this? Thank you.

StefanSchippers, 2022-09-23:
Please try to do:
`./configure --prefix=<install root> `
where `<install root> `is the top directory where you want to install xschem
some typical cases are:
`./configure --debug  --prefix=$HOME`
or
`./configure --debug --prefix=/usr/local`
Then do a `make && make install ` (1st case) 
or `make && sudo make install `(2nd case).

Don't set anything else with configure.

kwmartin, 2022-09-27:
I did as suggested; the make and install went properly. I then had to fix the permissions as my umask is 0027. I then called xschem from ~/.xschem and still got the error;
> cd /usr/local/share/xschem/
> ssu
# chmod 755 $(find . -type d)
# chmod 644 $(find . -type f)
# exit
> cd
> cd .xschem
> xschem
Tcl_AppInit() error: can not execute /usr/local/share/xschem/xschem.tcl, please fix:
can't read "USER_CONF_DIR": no such variable

Is USER_CONF_DIR an environmental variable or a macro definition or a global?

[image]

StefanSchippers, 2022-09-27:
USER_CONF_DIR is a preprocessor macro, it is hard coded (after ./configure) in the xschem source files. At runtime this macro is also copied to a TCL variable, before sourcing the xschem.tcl file.

Please check the config.h file in the top xschem source tree. It should contain a line liek this:

`#define USER_CONF_DIR        "~/.xschem"`

when running xschem, before reading the xschemrc ffle and the xschem.tcl file a tcl variable with the same name is created:
```
xschem [/mnt/sda7/home/schippes/xschem-repo/trunk/src] puts $USER_CONF_DIR
/home/schippes/.xschem
```

Please can you send me the `scconfig/config.log`, the `Makefile.conf` and the `config.h` files ?

StefanSchippers, 2022-09-27:
Also please ENSURE you don't have another xschem version installed on the system.
You can try `which xschem` to verify what is executed when you type `xschem`.

If needed do a hardcore search, it will take some time but it is worth doing:
find / -type f -name xschem

kwmartin, 2022-09-28:
From .git/config:
url = https://github.com/StefanSchippers/xschem.git

from:
grep -RnI 'USER_CONF_DIR' > grep_USER_CONF_DIR
file:///media/martin/ssd1/Downloads/IC_Tools/xschem/grep_USER_CONF_DIR

Re:
Please check the config.h file in the top xschem source tree. It should contain a line like this:
#define USER_CONF_DIR "~/.xschem"
yes it is.
> which xschem
/usr/bin/xschem
> 

file:///media/martin/ssd1/Downloads/IC_Tools/xschem/scconfig/config.log
file:///media/martin/ssd1/Downloads/IC_Tools/xschem/config.h
file:///media/martin/ssd1/Downloads/IC_Tools/xschem/Makefile.conf

re:
sudo find / -type f -name xschem
still running, nothing yet; I'll let you know.

I'll keep working on it and let you know if I come up with anything; very strange.

StefanSchippers, 2022-09-28:
If you did not specifically set `PREFIX=/usr` during configure it is very likely that a linux xschem package is installed.

try to do:
` dpkg -l|grep xschem`
if xschem is listed remove it:
`sudo apt purge xschem`

Stefan