StefanSchippers/xschem #84: Verilog arrays and `include

kwmartin, 2022-09-28:
Is it possible to declare a symbol as a primitive, but also to have a property with the name of a sym_name.v file found in one of the libraries which contains the verilog code for the symbol?
Similar to above, but a single file is loaded with all the modules for a verilog digital library?
Is it possible to to netlist verilog arrays as opposed to vectors? Such as memory arrays (section 3.10 of IEEE verilog standard)?

If the answer to some of the above is no (my expectation), could there be a simple way to implement it (especially for arrays - I'm guessing using an `include IEEE standard 19.5) statement can be used for 1 and 2?

StefanSchippers, 2022-09-29:
A single file containing library modules for a whole set of components is usually included in the parent as a
```
`include /path/to/lib.v
```
the idea of an attribute attached to a symbol containing the verilog file for it is interesting. 
I have created  `verilog_sym_def`  `spice_sym_def` and `vhdl_sym_def` attributes, if defined and not empty
the corresponding netlister will ignore the schematic (if any) of the symbol and just print the content of the attribute.
if you set this in the symbol:

```
verilog_sym_def="tcleval(`include \"[abs_sym_path verilog_include_file.v]\")"
```
the following lines will be inserted in the netlist:
```
...
...
// expanding   symbol:  verilog_include.sym # of pins=3
// sym_path: /home/schippes/.xschem/xschem_library/verilog_include.sym
`include "/home/schippes/.xschem/xschem_library/verilog_include_file.v"
``` 
As you can see the `verilog_include_file.v` has been fully qualified, the `abs_sym_path` tcl function searches for the file in any places defined in the `XSCHEM_LIBRARY_PATH` variable. Since a tcl function is used the entire attribute value is wrapped inside a tcleval(...)

For array pins/nets this is probably tricky. As far as i know even Cadence software did not allow multidimensional nets/pins.
However in xschem there is no assumption of a type for a wire/reg,you can declare a net of a custom type by setting verilog_type=... attribute, so may be you can have a vector of vector-typed pin/nets. I have to read the standard before.

kwmartin, 2022-09-29:
Thank you, I'll do a git pull, re-build and give it a try. Re the verilog arrays; I'm mostly interested in this for gates and modules and don't expect it will be possible; I have an experimental netlist for a simple minded logic simulator and I'm working on generating the experimental net list using  xschem. To give you feel, below is a D-ff in my netlist and you can see the gates are defined and connected using arrays. This probably doesn't fit in verilog very well. For now, I will without it and just use, for example, NOR nor3(2), nor2(2), nor1(2), nor0(2);
`
BEGIN CIRCUIT D_ff
    BEGIN DECLARATIONS
	INPUT in(D,CLK);
	OUTPUT out<Q,NQ>;
	NOR nor[4](2);
	AND and[4](2);
	INVERTER inv[2];
    END DECLARATIONS; 
    BEGIN CONNECTIONS
	inv[0] = in(D);
	inv[1] = in(CLK);
	and[0] = in(D),in(CLK);
	and[1] = in(CLK),inv[0];
	nor[0] = and[0],nor[1];
	nor[1] = nor[0],and[1];
	and[2] = nor[0],inv[1];
	and[3] = inv[1],nor[1];
	nor[2] = and[2],nor[3];
	nor[3] = nor[2],and[3];
	out<Q> = nor[2];
	out<NQ> = nor[3];
    END CONNECTIONS;
END CIRCUIT;
`

StefanSchippers, 2022-09-29:
what is the meaning of nor\[4\]\(2\)  ? I mean what does 4 and 2 mean ? 
Do you have a short document desribing the netlist format and syntax?
looks interesting. I can easily add new netlist backends.

You can play even with xschem commands
The following  tcl script prints all instances in circuit with all pin-->net associations:

```
# get number of instances
set inst [xschem get instances]
for {set i 0} {$i < $inst} { incr i} {
  # get symbol referenced by instance $i
  set component [xschem getprop instance $i cell::name]
  # get instance name followed by pin name and net name for each of instance pins.
  puts "$component [xschem instance_nodemap $i]"
}

```
You will get something like:
```
...
...
nmos4.sym m1 d #net1 g GN s 0 b 0 
pmos4.sym m2 d DIFFOUT g G s VCC b VCC 
vsource.sym VVCC p VCC m 0 
lab_pin.sym p1 p 0 
lab_pin.sym p2 p VCC 
nmos4.sym m3 d GN g GN s 0 b 0 
lab_pin.sym p3 p 0 
isource.sym IBIAS p 0 m GN 
lab_pin.sym p4 p 0 
...
...
```
one of the things that i have to do is to write a description of all these xschem query commands. It will go in the [developer section](https://xschem.sourceforge.io/stefan/xschem_man/developer_info.html) of the manual.

I think i have all commands to generate a custom netlist format with a tcl script (and if needed i can add missing query commands)

kwmartin, 2022-09-29:
1) Re: verilog_sym_def="tcleval(`include \"[abs_sym_path verilog_include_file.v]\")"
I'm in the process of giving this a try and from the figure, I noticed the symbol is a type=subcircuit rather than a verilog_primitive. Does this mean we need to include a "@symname.sch" file? (p.s. I 'm still having problems having seperate sym and sch directories - I'll figure this out soon - not sure how many levels your searching for files goes down).
2) Re the syntax of my experimental netlisting. It's from a long time ago and never saw the light of day, but I have been working (for fun) on bringing it back from the dead. The syntax NOR nor[4](2); declares 4 nor gates called nor[0], nor[1], .. to nor[3]. The "[4]" indicates we have 4 NOR gates; "NOR" is the type, "nor" is the name. The "(2)" indicates each gate has 2 inputs, and defaults to a single output. One could have also declared it as NOR nor[4](2)<1>; So round brackets are used for inputs, "<>" are used for outputs, and curly braces are used for inout buses (i.e. "{}). I do have pdf documents that describe the syntax, and a lot more (they were originally done in Framemaker that I no longer have access to and I haven't converted them to latex yet (or possibly Libre Writer), but I don't know how to store them somewhere so I can put a link to them into this comment. I guess I could make a github public repository and link to it; would this be your suggestion as the best way to go?

kwmartin, 2022-09-29:
Should have read `NOR nor[4](2); and NOR nor[4](2),1>`; the markup did something funny without the ``.

StefanSchippers, 2022-09-29:
For 1.  yes the symbol is of type subcitrcuit. This tells the netlister to look into it after netlisting the parent level. However if this new `verilog_sym_def` is defined there is no need to have a .sch file.

For the separate directories of .sch files they will be found if they are in directories listed in XSCHEM_LIBRARY_PATH.
Xschem does not do a full filesystem search or descend into subdirectories within these directories, this could take long time and lead to unexpected results.

kwmartin, 2022-09-29:
First try: working except for expanding abs_sym_path
set XSCHEM_LIBRARY_PATH /home/martin/.xschem/dig_lib/:$XSCHEM_LIBRARY_PATH:/home/martin/.xschem/std_dig_lib/:/home/martin/.xschem/proj_lib/:/home/martin/.xschem/xschem_library/examples/:/usr/local/share/doc/xschem/examples

The relevant include lines were (at the end of the netlist file in the global domain:

// expanding   symbol:  sym/dg_and2_2x1.sym # of pins=3
// sym_path: /home/martin/.xschem/std_dig_lib//sym/dg_and2_2x1.sym
`include "/home/martin/.xschem/xschem_library/devices/dg_and.v"

Note: /home/martin/.xschem/xschem_library/devices/ is not in XSCHEM_LIBRARY_PATH and also the symbols wasn't expanded correctly (it didn't include the "2_2x1" portion of the symbol name.

kwmartin, 2022-09-29:
Also, when I included in the top level shematic, the text:
`include "/home/martin/IC_Design/Models/INCLUDE/diglib28.v"
This was included in the  netlist inside the top-level module, whereas I think I need to figure out how to have it outside the top-level module at the global level?. I'm guessing if I make a symbol for the top-level and add verilog_sym_def = ..., this might do it? Is this what you might suggest?

StefanSchippers, 2022-09-29:
To include something at the global level there is a specific symbol for that: `devices/verilog_preprocessor.sym`
place that symbol and add the text you need. This will go out of any module scope.

StefanSchippers, 2022-09-29:
so in your tests with `verilog_sym_def` part of the filename gets truncated (no 2_2x1) ? this is very strange, i tested with a similar name:
```
verilog_sym_def="tcleval(`include \"[abs_sym_path verilog_include_file2_2x1.v]\")"
```
and result in netlist was correct:
```
// expanding   symbol:  verilog_include.sym # of pins=3
// sym_path: /home/schippes/.xschem/xschem_library/verilog_include.sym
`include "/home/schippes/.xschem/xschem_library/verilog_include_file2_2x1.v"
```

so I am interested in additional details so i can reproduce this strange behavior.

kwmartin, 2022-09-29:
It looks like the dropping of the _2x2 part was because I copied it from these comments and markup actually dropped it. I copied your line except for changing the symname to dg_and2_2x1 and now I'm getting:

// expanding   symbol:  sym/dg_and2_2x1.sym # of pins=3
// sym_path: /home/martin/.xschem/std_dig_lib//sym/dg_and2_2x1.sym
`include "/home/martin/.xschem/proj_lib/sch/dg_and2_2x1.v"

Note that /home/martin/.xschem/proj_lib/sch/ contains only the top level sch file, that is tstDig2.sch; so it looks to me that maybe the abs_sym_path is expanding to the directory where the top level sch is, not the directory the symbol is in. Maybe we need a means of specifying a directory for saving all the verilog files in?
I have also simplified the example and can send it to you; which would be the simplest way to give it to you?

StefanSchippers, 2022-09-29:
Note that in my example I used tcleval() and [abs_sym_path ...] just to let xschem find the file in one of the XSCHEM_LIBRARY_PATH directories and build the full path for it.
You can make everything simpler and faster (but less portable to a different system) by specifying the path:
```
 verilog_sym_def="`include \"/path/to/verilog_include_file.v\""
```

the abs_sym_path returns the path of the file if found in one of the XSCHEM_LIBRARY_PATH directories.
If nothing is found it returns the path of the containing schematic.

StefanSchippers, 2022-09-29:
for the example pack everything into a tar.gz file and send to me  stefan.schippers@gmail.com

StefanSchippers, 2022-09-29:
another comment. If in your top schematic you instantiate a symbol reference as  `sym/dg_and2_2x1.sym`, you should give 
`sym/dg_and2_2x1.v` as argument to the abs_sym_path function, so in the symbol attributes:
```
verilog_sym_def="tcleval(`include \"[abs_sym_path sym/dg_and2_2x1.v]\")"
```

another note, i also had issues with the markdown eating characters here , when pasting text with special characters
start a line with 3 characters like this:
\`\`\`
past the text after and then close with another 3 characters:
\`\`\`

kwmartin, 2022-09-30:
Things are coming; I'll send you my example once I get a reasonable version together, which is happening. Quick question, in one of your videos (I'm working through all of them) you mention that one of your videos deals with "extra" pins and how to use them. Can tell me which video? Also, your explanations above about abs_sym_path really help me understand what is happening.

StefanSchippers, 2022-09-30:
abs_sym_path is also the internal function used by xschem to locate schematics/symbols.
so if you place in a schematic a symbol reference as `sym/xxx.sym` and your XSCHEM_LIBRARY_PATH is a list of paths:
```
/first/path:/second/path:/third/path
```
then the following places will be examined for the symbol:
```
/first/path/sym/xxx.sym
/second/path/sym/xxx.sym
/third/path/sym/xxx.sym
```
If no symbol is found it is assumed to be:
```
[directory of parent schematic]//sym/xxx.sym
```
The only reason for this last resort choice is to make things work for beginner users with garbled search path.

StefanSchippers, 2022-09-30:
For the 'extra' question, it is about symbol pins that are defined via attributes. [This is the video](https://xschem.sourceforge.io/stefan/xschem_man/video_tutorials/inherited_connections.mp4).

kwmartin, 2022-09-30:
A related question, is it possible to make pin labels invisible or alternatively, have a very small font size?

StefanSchippers, 2022-09-30:
For the extra pin giving parameters in verilog i need to check, i think you found a bug.
Will fix
Spice netlisting always requires a format attribute, for primitive as well as subcircuits, since the resulting lines have the very same format.
Verilog and VHDL only require verilog_format and vhdl_format for primitives (for example a switch or a nmos in verilog)
for subcitcuits the calling convention is done automatically with the correct syntax required by the language, so in this case no verilog_format or vhdl_format must be present as it overrides the standard instance calling convention

StefanSchippers, 2022-09-30:
Not really a bug, simply the extra stuff is not implemented for verilog. WIll add that, so the netlists will be consistent. Will take some time, i think will be done tomorrow.

kwmartin, 2022-09-30:
Re traditional order based verilog netlisting:
verilog_format="@symname @name ( @pinlist );"
produces a line like:
dg_nand2_2x1 nand0 ( d clk net4  { VCC,VSS}    );
This is syntatically incorrect; separating commas are missing in the bus specification; it should be:
dg_nand2_2x1 nand0 ( d, clk, net4,  { VCC,VSS}    );

StefanSchippers, 2022-10-01:
the @pinlist is mostly used for cpice, it expands to the list of pins in the symbol in the order they were created (or adjusted using the Shift-S 'change order' command) and separated by spaces.
You can get far more control by using the @@ syntax.
@@A will expand in netlist to the net attached to node A of the symbol.
you can use in above example:
verilog_format="@symname @name( @@d , @@clk , @@q , <additional @params if needed> );"

StefanSchippers, 2022-10-02:
I have fixed the verilog netlister regarding to pins listed in the new 'verilog_extra' attribute for verilog subcircuits, that do not have a verilog_format attribute . Mostly used for logic cells where power connections are usually the same for all cells so creating pins for all of these is a hassle. Moreover, having power pins passed by attribute allows you to block-select all standard cells (perhaps with a 'search function') and a single 'q' allows you to re-route the power connections. In verilog netlists the extra attribute is still used together with verilog_extra. Ths first contains all attributes NOT to pass as component/symbol parameters, the second defines the list of attributes to use as pins. Since components may have more attributes than those needed for verilog (for example specific spice attributes) I keep these two attributes separate. The extra is an exclude-list for parameters, the verilog_extra is an include-list for pins.
Documented in the [manual](https://xschem.sourceforge.io/stefan/xschem_man/symbol_property_syntax.html): 

The following symbol, with these attributes:

```
type=subcircuit
format="@name @@A @@B @VGND @VNB @VPB @VPWR @@Y @prefix\\mynor2"
template="name=x1 VGND=VGND VNB=VNB VPB=VPB VPWR=VPWR prefix=spicemodel_"
extra="VGND VNB VPB VPWR prefix"
verilog_extra="VGND VNB VPB VPWR"

```

will be used in verilog netlists as follows:
```
// sch_path: /home/schippes/.xschem/xschem_library/test_mynor2.sch
module test_mynor2
(
  output wire YY,
  inout wire VSS,
  inout wire VCC,
  input wire BB,
  input wire AA
);
mynor2
x1 ( 
 .A( AA ),
 .B( BB ),
 .Y( YY ),
 .VGND( VSS ),
 .VNB( VSS ),
 .VPB( VCC ),
 .VPWR( VCC )
);

endmodule

// expanding   symbol:  mynor2.sym # of pins=3
// sym_path: /home/schippes/.xschem/xschem_library/mynor2.sym
// sch_path: /home/schippes/.xschem/xschem_library/mynor2.sch
module mynor2
(
  input wire A,
  input wire B,
  output wire Y,
  inout wire VGND,
  inout wire VNB,
  inout wire VPB,
  inout wire VPWR
);
endmodule

```

closed 2022-12-13 by StefanSchippers