StefanSchippers/xschem #85: Issue with running ngspice in xschem

ChrisZonghaoLi, 2022-09-29:
Greetings,

So I was following Stefan's videos (https://www.youtube.com/watch?v=jXmmxO8WG8s) to get Ngspice (37) installed and I also have Xschme and sky130 pdk installed (https://www.youtube.com/watch?v=KgBLByOkJxA). I tried to run the testing examples in Xschme (with all setup done by strictly following your tutorials) by clicking the simulation bottom, but there was never a window popped out. And in the command window it just showed: 

~~~
% Simulation started: execution ID: 0
~~~

It was fine in the batch mode since I can confirm that the simulation indeed runs (get .raw files), but not the case with interactive mode ($terminal -e 'ngspice -i "$N" -a || sh').

A quick note is that there is a probability that I missed install something very primitive. Since my system is very clean, before I could not even open the netlist ("Simulation->Edit Netlist"), and no error popped out whatsoever in the command window. And it was solved after I installed gvim on my OS. 

I am using Ubuntu 20.04.

Thanks!

Chris

StefanSchippers, 2022-09-29:
for interactive simulations xschem launches a terminal and inside the terminal ngspice is run. The default setting is to use `xterm`. You can instal `xterm` by doing` sudo apt install xterm`. You can also edit your `~/.xschem/xschemrc` file and change the terminal tcl variable. Search for 'terminal and read comments.

You can add this line:
`set terminal gnome-terminal`
to use the gnome default terminal. 
My suggestion is to use xterm, it is more robust and is a very small application.

ChrisZonghaoLi, 2022-09-29:
This is great! Thanks Stefan! You have been very helpful!

closed 2022-09-29 by ChrisZonghaoLi