StefanSchippers/xschem #86: Small Bug

kwmartin, 2022-10-03:
I'm often seeing the first letter of sym being dropped when, for example, doing a <shift>I or a save as. Here is an example
[image]
Note the second line starts with ym rather than sym. When I add the missing s, the symbol is then found.

StefanSchippers, 2022-10-03:
does **/home/martin/.xschem/proj_lib/sch/ym** directory exist?

StefanSchippers, 2022-10-10:
Check all the xschemrc files lying around. 

- system xschemrc is in <prefix>/share/xschem/xschemrc. This usually is all commented, so no setting is given at global level.
- check if there is an xschemrc file in the directory from where you launch xschem
- if no xschemrc is found in current directory then xschem  tries to load ~/.xschem/xschemrc

SInce the system-wide xschemrc is by default empty, xschem usually loads only one xschemrc file, the one in current dir or (if not found) the one in user home: ~/.xschem/xschemrc.

When setting your search paths in your xschemrc **ALWAYS** begin with a:
set XSCHEM_LIBRARY_PATH {}
then:
append XSCHEM_LIBRARY_PATH :/path/to/xschem/lib
...

this will clear any previous definition if any.

i see some paths are repeated, 

also note that if you run xschem in the build directory (src/) then a local setup is used (use all symbol directories from src/xschem_library), this is used to test new xschem versions.

If no assignment is made to XSCHEM_LIBRARY_PATH (because no xschemrc files loaded)  then this list of paths is used  (of course with the /home/schippes/share/ replaced with your install prefix):

#define XSCHEM_LIBRARY_PATH  "~/.xschem/xschem_library:/home/schippes/share/xschem/xschem_library/devices:/home/schippes/share/doc/xschem/examples:/home/schippes/share/doc/xschem/ngspice:/home/schippes/share/doc/xschem/logic:/home/schippes/share/doc/xschem/xschem_simulator:/home/schippes/share/doc/xschem/binto7seg:/home/schippes/share/doc/xschem/pcb:/home/schippes/share/doc/xschem/rom8k"

closed 2022-12-13 by StefanSchippers