StefanSchippers/xschem #87: Verilog Netlisting with Parameters

kwmartin, 2022-10-08:
If I have the following for my symbol parameters for a 2-input nor gate:
'''
type=verilog_primitive
format="@name @pinlist @VCCPIN @VSSPIN @symname"
xxverilog_format="@symname #( @circuit ) @name ( @pinlist , @VCCPIN , @VSSPIN );"
template="name=nor1 circuit=async VCCPIN=VDD VSSPIN=VSS"
extra="VCCPIN VSSPIN"
verilog_extra="VCCPIN VSSPIN"
generic_type="circuit=string"
'''
I get following in the netlist:
'''
dg_nor2_2x1
#(
.circuit ( "async" )
)
nor1 ( 
 .A( nr0 ),
 .B( a1 ),
 .Z( nr1 )
);
'''
Note: a) there are quotes aroung "async" and b) no pins VDD and VSS.
If I use the following for my symbol parameters:
'''
type=verilog_primitive
format="@name @pinlist @VCCPIN @VSSPIN @symname"
verilog_format="@symname #( @circuit ) @name ( @pinlist , @VCCPIN , @VSSPIN );"
template="name=nor1 circuit=async VCCPIN=VDD VSSPIN=VSS"
extra="VCCPIN VSSPIN"
verilog_extra="VCCPIN VSSPIN"
generic_type="circuit=string"
'''
where now verilog_format is active, I get:
'''
dg_nor2_2x1 #( async ) nor1 ( nr0 , a1 , nr1 , VDD , VSS );
'''
We now have the VDD and VSS pins, but async is not surrounded by quotes, that is, it should be "async". I tried
verilog_format="@symname #( \"@circuit\" ) @name ( @pinlist , @VCCPIN , @VSSPIN );"
and that gave:
dg_nor2_2x1 #( " ) nor1 ( nr0 , a1 , nr1 , VDD , VSS );

What I think is required is:
'''
dg_nor2_2x1 #(  "async"  ) nor1 ( nr0 , a1 , nr1 , VDD , VSS );
'''

kwmartin, 2022-10-08:
Uh, markup issue: should have read, I tried:
```
verilog_format="@symname #(  \"@circuit\"  ) @name ( @pinlist , @VCCPIN , @VSSPIN );"
```

StefanSchippers, 2022-10-09:
if you specify verilog_format you have full conrol of the output, so the generic_type is irrelevant.
Use this:
```
verilog_format="@symname #(  \"@circuit\\\"  ) @name ( @pinlist , @VCCPIN , @VSSPIN );"

```

if there are no spaces after a @parameter a backslash tells xschem the end of the pareter name. since everything is inside quotes a double backslash is needed.

StefanSchippers, 2022-10-09:
in the above code for the nor2 symbol, if verilog_format is commented out (as in xxverilog_format) the VDD and VSS pins are printed in netlist if the instance has attributes VCCPIN=VDD VSSPIN=VSS.

I have updated xschem so now if no attributes are set to the instance these will be taken from the symbol template string.
This is the general way for taking default values for parameters (and extra pins like in this case) if these are not given as instance attributes. It did not work for verilog netlist due to a typo in the code. Thank you for pointing the issue out.

StefanSchippers, 2022-10-09:
please also note there is no `type=verilog_primitive` (although for your examples anything different from `subcircuit` will work).
a `type=primitive` will cause all netlister formats to consider the symbol a terminal symbol and avoid descending into its schematic. A symbol of type `primitive` allows to descend (with the usual xschem commands, for example 'e' key) into its schematic if any, but it will not be used in any netlist format. A type that is different from `subcircuit` or `primitive` will not allow to descend and view the schematic even it there is one.
you can set a symbol to be a primitive only for verilog, while letting the spice netlister descend into its schematic by setting:
```
type=subcircuit
verilog_primitive=true
```