StefanSchippers/xschem #88: Flatten netlist with new components

cardosorapha, 2022-10-09:
Hi, 

I intend on using xschem as a schematic editor for photonic circuits, and I already created new base symbols that are correctly transformed into a netlist. 

I am trying to deal with hierarchy, as the flattenning script (flatten.awk) is tied to the common electronic devices, as I can see from this declaration:

BEGIN{
 # topcell=toupper(ARGV[2])
 # ARGC=2
 first_subckt = 1
 pathsep="_"
 nodes["M"]=4; nodes["R"]=2; nodes["D"]=2; nodes["V"]=2
 nodes["I"]=2; nodes["C"]=2; nodes["L"]=2; nodes["Q"]=3
 nodes["E"]=4; nodes["G"]=4; nodes["H"]=2; nodes["F"]=2
}

What I got from trying to use it was the following:
    - A deviced that is called DC1 with 4 nodes was correctly flattened to its first two nodes. So I associated it to the presence of the nodes["D"]=2 declaration. When I changed it to 4 it correctly parsed its 4 nodes in the hierarchy. 
    -  A deviced that is called WG1 with 2 nodes was included in the netlist but without new identifiers to its nodes, denoting the subcircuit it comes from. I tried to include nodes["W"]=2 but it did not work. 

The question then is: is there a simple way to include my new devices in the context of xschem's flattenning script?

StefanSchippers, 2022-10-09:
you probably modified the flaten.awk in the src directory
you need to add the nodes["W"]=4 as you did but also add W in the following line: 
```
   if(line[1] ~ /^[WEGHFCMDQRGIV]/)
```

also remember to do a 
```
make install 
```
if you do the change in the src/ directory.

cardosorapha, 2022-10-09:
Thank you very much for the response! I just tried it and it works just fine.

It was surprising that in a day's work of reading documentation and drawing symbols I have an open source photonic schematic editor with netlisting (attached a picture of a simple schematic)! Thanks for the work on this project. 

[image]

Making it general would facilitate any expansions I would say, since we wouldn't be limited to one letter per pin count. Although I could work around it with a convention like A=1 node, B=2 nodes, ...

StefanSchippers, 2022-10-10:
I have a question. is there a rule to get the name of pins of new components?
In spice netlists an element line may look like:

`Wxxxx node1 node2 ...  model par1=val1 par2=val2 ...`
In some cases element lines contain a reference to a model, in other cases they don't.
It is difficult to decide in example above if `model` is a net connection or not.

StefanSchippers, 2022-10-10:
I have updated xschem so to define a new component 'W' you just need to provide the `nodes["W"]=2` (assiming 2 is the # of pins).
The line: 
` if(line[1] ~ /^[WEGHFCMDQRGIV]/)`
is now replaced with:
`if(line[1] ~ devpattern)`
where `devpattern` is obtained automatically from the `nodes[]` entries, so no change is required here anymore.

Next change will be to automatically generate the nodes[] array for new elements, once i know what is the rule to get the number of pins.

StefanSchippers, 2022-10-10:
I have added a simple way to specify custom devices pin count to the netlist flattener.
Just add in your schematic this line: 
*.nodes[W]=2
or:
*.nodes["W"] = 2
You can do this in your top level netlist by pressing 'q' with no elements selected in the schematic and inserting above text.
[image]

If you have time please update your xschem and test.
Thank you

cardosorapha, 2022-10-10:
Oh, thank you again ! I just tested the modified flatten.awk and it works just fine with new component nodes in the schematic properties. I included all my new devices and they were all correctly flattened in the netlist. 

About your question above: indeed, it is not easy to infer the number of nodes just from the parameters that have no "param=xx" because of things like MOSFET models. A suggestion then: what if the script fetched the number (N) of pins from the component symbol? Then it could assume that the N first strings after its name are net names.

StefanSchippers, 2022-10-11:
Just curious. 
Why do you need a flat netlist? are you using some special simulator that does not handle hierarchy?

cardosorapha, 2022-10-11:
Exactly. We're still developing the simulator based on reading spice netlists, but we can't handle hierarchy yet. As I found that xschem flattens I figured we can focus on development of more backend while xschem stays in the front end.

kwmartin, 2022-10-12:
Sort of related. In the simulators I am playing with, I have a single simulation function for say all nand gates, and then the function figures out how many inputs, and I have parameters for gate strength (i.e. output admittance) and output capacitance. So, a wish list: one symbol that can be used for all nand gates irrespective of how many inputs; the number of inputs is specified at instantiation time? As opposed to a symbol for 2 input nand gates, a different symbol for 3 input nand gates, and a third symbol for 4 input nand gates? Not an important issue, it could just help to keep libraries smaller.

StefanSchippers, 2022-10-12:
I often use parametrized blocks for VHDL of Verilog. Below example is a ROM macro, with variable width address and data bus:
[image]

the corresponding schematic uses 'dim' and 'width' for the bussed ports:
[image]

StefanSchippers, 2022-10-12:
@cardosorapha 
A flat netlist is not generated directly by xschem since xschem is inherently a hierarchical and parametric schematic editor. While (expecially way back when I started developing it) there are many flat schematic editors that do some 'hacks' to allow hierarchy and subcircuits (and in most cases without passing parameters), for xschem we have the opposite case. Some post processing converts a hierarchical netlist into flat.

The flatten.awk script can be used as standalone: ./flatten.awk netlist.spice > flat_netlist.spice

So any information about new or custom devices should be passed as metadata into the netlist.
A common approach is to use the `*.` pattern to begin a line. The `*` marks the line as a comment so it will be ignored by the simulator, however the `.` (which is usually a spice command line, like for example `.tran`) after the `*` will be used by some other backend tool that prepares the netlist for final simulation.
Cadence software uses ofthen this metadata trick to add pin direction information into spice netlists, like: 
```
.subckt comp_ngspice PLUS MINUS OUT
*.PININFO PLUS:I MINUS:I OUT:O
...
...
.ends
```

cardosorapha, 2022-10-17:
I understand. After using it for a while for a few tests, I'm still taking xschem as my schematic editor of choice for the simulator we're developing. I'll put in the todo list to handle hierarchy inside of it but while it's not there the flatten.awk script will to wonders to allow circuit investigation. 

Thank you again for the support and the quick responses.

closed 2022-10-17 by cardosorapha