StefanSchippers/xschem #89: Make Symbol From Schematic

kwmartin, 2022-10-13:
When I do this command at the my top level test bench (with nothing selecte but containing In and Out pins), a Dialog pops up asking for confirmation. After confirming, I can't find the new sym file. Is there a log file in case I did something wrong, or a default location for the file in case it was successfully created?

StefanSchippers, 2022-10-14:
The symbol is generated by taking the full absolute path of the schematic, creating a new filename with trailing .sch replaced with .sym and writing the symbol into that file.

In other words the symbol is created in the same directory of the schematic.
If you can't find it there, there is some problem and the symbol generator did not complete for some reason.

I have done two tests. First by deliberately inserting a syntax error into the make_sym.awk symbol generator, the second by removing the make_sym.awk file. xschem reports these two errors into the terminal it is launched from:
```
schippes@mazinga:~$ xschem
xschem [~] tcleval(): evaluation of script: make_symbol {/home/schippes/.xschem/xschem_library/test spaces.sch} failed
         : awk: 34: unexpected character '`'

schippes@mazinga:~$ xschem
xschem [~] tcleval(): evaluation of script: make_symbol {/home/schippes/.xschem/xschem_library/test spaces.sch} failed
         : awk: cannot open /home/schippes/share/xschem/make_sym.awk (No such file or directory)
```

kwmartin, 2022-10-14:
Thank you, I verify the symbol was generated in the same directory as the schematic. I had opened the schematic using File->Open recent, and I mistakenly thought the schematic was in a different directory.

closed 2022-10-14 by StefanSchippers