StefanSchippers/xschem #90: Possible Suggestion

kwmartin, 2022-10-14:
As a default, add another parameter to symbol pins called pinnumberr, starting at 0. This could be used in verilog netlisting style where the pin ordering is used. It shouldn't change things as currently I think pin ordering is decided by the order the pins are added. The additional pinnumber attribute would a) make this order obvious at a later date, and b) allow the order to be controlled and easily changed. Being honest, it's a change that would personally save me a bit of time, and might not be much use to the majority of others. I do think it's closer to what Viewdraw does? I'm guessing you have a prioritised list of possible changes and this might be low down.

StefanSchippers, 2022-10-14:
I can do that. Please consider that if you generate a symbol automatically from the schematic the vertical position of schematic I/O pins determines the order. In below example the symbol will have the OUT pin at first position (0) then the input pins follow in the vertical order they are drawn in the schematic.

[image]

Netlist of the symbol expansion will look like:
```
.subckt mos_power_ampli OUT MINUS PLUS VSS VPP VNN
*.ipin PLUS
*.ipin VPP
*.ipin VNN
*.opin OUT
*.ipin MINUS
*.ipin VSS
...
...
.ends
```
If I add a pinnumber attribute can I start from 1? All pcb softwares use pinnumber to determine the pin position on the package, and these numbers start at 1.

StefanSchippers, 2022-10-16:
If all symbol pins have a pinnumber attribute this symbol will be netlisted (in all netlist formats) with pins sorting in ascending order according to pinnumber value. Start value of pinnumber does not matter (may start at 1 or 0) , it is used as the sort key.

You can assign the pinnumber attribute directly in the symbol:

[image]

or you can assign these in the schematic pins, if you use the "Make symbol from schematic" function ('a' key) these attributes will be transferred to the symbol.

[image]

closed 2022-12-13 by StefanSchippers