StefanSchippers/xschem #91: TCP Connection Doesn't Work with -x

yrrapt, 2022-10-24:
I have `set xschem_listen_port 2021` in `xschemrc` to use a TCP connection.

When I start xschem using the command `xschem` everything works correctly. I can use `nc localhost 2021` to communicate with xschem.

However, when I start xschem without GUI using `xschem -x` the connection with `nc localhost 2021` hangs.

Is this a bug or expected behavior?

Thanks in advance,
Thomas

StefanSchippers, 2022-10-24:
I have the same behavior. Will look into it.
Thanks for now.

StefanSchippers, 2022-10-24:
Thomas i think it is working now.

The issue is that if xschem is run with no X (-x) and with no tclreadline (if you have this package installed, -r) no event loop is active. to let xschem accept tcp connections you must suspend the xschem command shell, for example with:
vwait xxx
this command will wait until xxx is written. While suspended xschem will service tcp requests.

I have now added a tcl event handler if xschem is run with -x option (and -r on systems with tclreadline package).

From the tcl manuals, if a tcl event handler is installed the prompt itself will become event based and prompt will be serviced if stdin becomes readable.

so what i had to do was to add this simple function:
static void tclmainloop(void)
{
  while(1) Tcl_DoOneEvent(TCL_ALL_EVENTS);
}

This event handler is installed only if no X and no tclreadline is used since in these cases an event loop is already provided.

Can you please update your xschem and test if everything is ok now?
Thanks