StefanSchippers/xschem #92: Installation error

realsurgent, 2022-10-29:
When I run "make -j4" I get the following error and can't install xschem.  
[image]

StefanSchippers, 2022-10-29:
You probably ran ./configure --prefix=/home/schippes.
You must use your username.

realsurgent, 2022-10-29:
I did what you suggested but it didn't work. But I did run setup from ` sudo apt-get install xschem`.

StefanSchippers, 2022-10-29:
Do not install xschem from the ubuntu app store with apt-get. the packaged xschem that comes with Ubuntu/Debian/Fedora is too old.

remove the xschem package: `sudo apt-get purge xschem`
Install xschem by  cloning the sources, follow the manual [here](https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html).

realsurgent, 2022-10-29:
> Do not install xschem from the ubuntu app store with apt-get. the packaged xschem that comes with Ubuntu/Debian/Fedora is too old.
>
> remove the xschem package: `sudo apt-get purge xschem` Install [...]

I did your suggestion and I still get the same error. Also xschem I downloaded from the app store works as I want.

StefanSchippers, 2022-10-29:
please post the steps you did from git clone to make install. Also attach the xschem_git/src/scconfig/config.log
xschem from ubuntu appstore is too old and you will get in trouble sooner or later.

remove the packaged xschem with sudo apt-get purge xschem, remove the source xschem directory, follow every step in the [manual](https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html).

It is impossible that make install tries to install in /home/schippes since you are on a different username, unless you typed schippes somewhere.

realsurgent, 2022-10-30:
I did it. All I had to write was `sudo make install`

closed 2022-10-31 by StefanSchippers