StefanSchippers/xschem #101: Unable to run xschem

yashmithak19, 2023-02-22:
Hi Team,

I am a 1st time user to xschem. I have followed all the steps for the configuration, make make and install as per the user manual and is successfully completed and the xschem.exe is available in the src folder. However on running the xschem.exe file, the xschem window doesn't open as expected. No error/warning message is generated.

 Could you please help me with this?

Note: In am accessing the LINUX environment and executing the commands using **Cygwin**
Attached are the snapshots for your reference.

[image]

[image]
[image]

StefanSchippers, 2023-02-22:
Xschem is running (you see the xschem tcl prompt %)and this is good. Xschem uses the xorg server to display graphics (like most unix graphical programs). You need to set up [Cygwin/X](https://x.cygwin.com/docs/ug/setup.html) and verify you can run simple X clients, like xclock, xterm and so on. (My suggestion is to install cygwin packages gvim, xterm and x11-apps). If you get these clients running you will be ok with xschem as well.
I did this some years ago. You can configure the cygwin Xserver to run in rootless /seamless mode, so the x apps look like native windows programs.
https://a.fsdn.com/con/app/proj/xschem/screenshots/xschem_on_windows.png/max/max/1

yashmithak19, 2023-02-24:
Your suggestion was really helpful. 
I already had the X11 packages installed. However since it was invoked incorrectly the xschem GUI was not working. 
On configuring cygwin Xserver to run in multiwindow mode and then invoking cygwin terminal from the System tool and running the xschem from this terminal worked.

[image]

Thank you for the support. really helpful!!

closed 2023-02-24 by yashmithak19