StefanSchippers/xschem #104: Adding tag for each release

atorkmabrains, 2023-05-18:
@StefanSchippers,

Thanks for great tool. I'm releasing a build system that generates an environment for work that is capable of having multiple versions of the same tool for appropriate usage. 

Having a version in the main repo of the tool is very helpful. Could you please add versions to your tool with every major release?

StefanSchippers, 2023-05-21:
xschem has a version that can be queried with xschem -v.

I am preparing for a new release **3.4.0**, that follows 3.1.0 done on Aug. 2022
If all tests are OK I will tag a new release.

There is no 'major release' development model in xschem, it's more like a continous development, with feature additions/bug fixes. But I agree with you it's time to tag a new point in the development.

atorkmabrains, 2023-05-21:
@StefanSchippers It would be nice to automate this tagging process using 'pre-commit'

Blebowski, 2023-06-29:
Hi @StefanSchippers, adding a tag more frequently (e.g. monthly) would be extremely usefull.
In a shared network setup, where one might need to have multiple versions of the tool installed
it is often difficult to figure out "which version is the latest". It is impossible to recognize latest
version only from GIT hash.

StefanSchippers, 2023-06-29:
I have tagged version 3.4.1. will do more frequent tags in the future.

closed 2023-08-04 by StefanSchippers