StefanSchippers/xschem #106: use VHDL/VERILOG tb in Xschem

Samarthjainabout, 2023-06-03:
I wanted to write a TB in verilog to check a digital circuit in Xschem. As an example, I saw this attached one but it did not run. Could you tell me how I can write a TB in xschem using Verilog?Any video or tutorial?
[image]

StefanSchippers, 2023-06-07:
You are setting VHDL netlist, while this is a Verilog example.

Ensure you have Icarus verilog (`iverilog`) and `gtkwave` installed.
Start xschem on the testbench
Go to `Options` and set `Verilog netlist`.
Press Netlist button, then Simulate button, then Waves button.
See attached video clip
https://github.com/StefanSchippers/xschem/assets/69359491/b7547d32-7b98-4155-b762-da5108a4121c

StefanSchippers, 2023-06-29:
Closing since no feedback. Feel free to reopen it if you have any news.

closed 2023-06-29 by StefanSchippers