StefanSchippers/xschem #109: Crache after wrong use of --script

aschauerf, 2023-06-19:
xschem starts and works after starting with wrong command-line options:

xschem --script $ASIC_BASE_DIR/open_pdks/sky130/sky130A/libs.tech/xschem/xschemrc

(correct:
xschem --rcfile $ASIC_BASE_DIR/open_pdks/sky130/sky130A/libs.tech/xschem/xschemrc
)

but a Segmentation fault (core dumped) occurs when a schematic is tried to be opened or saved.

StefanSchippers, 2023-06-19:
Thank you for this report. will investigate and add the necessary guards

StefanSchippers, 2023-06-19:
It is now fixed, was a regression due to changes done in 89d90c21dba1e70f39aaa1058b1e3e1e0436a6a9
If you have a chance to test please do it. 
Using --script for an rcfile will not prevent xschem to read the standard xschemrc file (if existing) in ~/.xschem/xschemrc, in this case you end up starting xschem with 2 xschemrc files sourced. This could lead to a mix of different paths or settings, but program should behave normally and not crash.
--rcfile for loading an xschemrc file is a better option since it prevents xschem from loading any other (default) xschemrc file

closed 2023-06-29 by StefanSchippers