StefanSchippers/xschem #110: The `Move` shortcut key causes the `object` to drift a long distance.

TuringKi, 2023-07-06:
[image]

When using the move shortcut key (`M`), the selected target does not move near the current mouse pointer, but drifts outside the canvas. This is shown in the above gif (may need to be enlarged to see clearly).

My OS Version:

```
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

xschem version: commit 587a847722e21c55c46c9250393aba5e36f852c1

TuringKi, 2023-07-07:
> In can not reproduce this here: https://github.com/StefanSchippers/xschem/assets/69359491/1cd404b4-7e33-4554-b06e-d1835816a95d what graphic card are you using? Are you using some remote graphic connection (VNC or similar)? does this happen also with move ('m') key?

Nvidia gpu 1080.

Not use remote connect.

`M` has the same issue.

StefanSchippers, 2023-08-04:
You should try to run the xserver using the fbdev device. Create a custom `xorg.conf` file and add `Driver "fbdev" `instead of `"nvidia"`. See if there is some difference. You can also start x11 using the open source `"nouveau"` driver. If you see correct behavior then the problem is in the nvidia graphic driver.

olisnr, 2023-09-20:
this shifted device i saw also whit the intel driver on UBUNTU 20.04

StefanSchippers, 2023-10-28:
Hi,it's beena long time, However I have implemented a fix that seems working on some systems that had a similar misbehavior. You can enable the fix in Options -> Fis broken RDP mouse coordinates.
[image]

TuringKi, 2023-10-28:
> Hi,it's beena long time, However I have implemented a fix that seems working on some systems that had a similar misbehavior. You can enable the fix in Options -> Fis broken RDP mouse coordinates. [image]

Thanks for fix, i will test it and give you feedback :)

closed 2023-12-11 by StefanSchippers