StefanSchippers/xschem #111: Expanded mosfet instance names can be duplicated in the schematic without a warning.

d-m-bailey, 2023-08-03:
Expanded mosfet instance names can be duplicated in the schematic without a warning.
mpin_fast[1]_q0 (green) will conflict with the expansion of mpin_fast[1:0]_q0 (magenta).
The spice netlist is valid, because the expanded mpin_fast[1]_q0 has been renamed to mpin_fast[1] (red).

Currently, it is impossible to add a simple instance name that is a duplicate of an existing instance name. Is it possible to extend that check to the expansion of array instances?

[image]

StefanSchippers, 2023-09-25:
@d-m-bailey after a long time i have addressed this long standing issue.
If you add a component its name will be checked for uniqueness.
the name is expanded to single bits, like V1[1:0] --> V1[1],V1[0]
and every single bit name is checked against all single bit names in the schematic. If there are duplicates the component is renamed, for example V1[3:0] --> V2[3:0] until all bits of the renamed instance are unique w.r.t all bits of all other instances.

[image]

d-m-bailey, 2023-09-25:
Excellent!

StefanSchippers, 2023-10-05:
I believe this is fixed so closing. Feel free to reopen if any problems found.

closed 2023-10-05 by StefanSchippers