StefanSchippers/xschem #112: Only [] are allowed for buses and arrays.

d-m-bailey, 2023-08-03:
Is it possible to add other punctuation to buses and arrays? 
Such as
```
a<0:5>
Xint{3:4}_d
OUT(8:9)
```

Currently, `<`,`>` doesn't expand.
```
Xmpin_fast<1:0>_q0 PD_I2C_H DRVLO_H_N net1 VCC_IO sky130_fd_pr__pfet_g5v0d10v5 L=1 W=0.42 nf=1 ad='int((nf+1)/2) * W/nf * 0.29'
+ as='int((nf+2)/2) * W/nf * 0.29' pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)'
+ nrd='0.29 / W' nrs='0.29 / W' sa=0 sb=0 sd=0 mult=1 m=1
```

d-m-bailey, 2023-08-04:
Related to array expansion of instance names, sometimes the suffix get dropped.
For example, `XI[3:0]_q` is expanded into `XI[3], XI[2], XI[1], XI[0]_q.`
[image]

StefanSchippers, 2023-08-04:
Expansion works only with [...] at the end of a node name.

a[2:0]_q expands to:
a[2],a[1],a[0]_q
a_q[2:0] expands to:
a_q[2],a_q[1],a_q[0]

Currently bus notation uses only [ and ] characters in schematic.
You can remap these character in produced netlist bu using menu
`Options->Replace [ and ] for buses in SPICE netlist"`
or setting a variable in the xschemrc file:
` set bus_replacement_char {<>}`

I will look if there is an easy way to modify the grammar parser to expand stuff like:
`a[2:0]_q`
into 
`a[2]_q,a[1]_q,a[0]_q`

d-m-bailey, 2023-08-10:
> You can remap these character in produced netlist bu using menu
> Options->Replace [ and ] for buses in SPICE netlist"
> or setting a variable in the xschemrc file:
>  set bus_replacement_char {<>}
> 
> It is not easy to add other characters (like (..) or <...>) for bus notation since these new characters become 'special' and performing bus expansion is not always what we want. Libraries for IC symbols use almost any available ascii character.

Thanks, @StefanSchippers. 
I currently don't have xschem compilation set up in my flow, but I'll look into it and let you know the test results.

The problem I ran into was with reverse engineering the sky130 gpio cell. There `<` `>` is used as the bus delimiter.

Incidentally, CVC recognizes any of `<>`, `[]`, `()`, `{}` as possible bus combinations because it has to deal with netlists from a variety of possibly mixed sources. Here's the [code](https://github.com/d-m-bailey/cvc/blob/master/src/CCvcDb_utility.cc#L815) for reference.

StefanSchippers, 2024-02-22:
Closing for now. 
If a CDL /SPICE netlist uses < ,> or other bus index delimiters the best way to deal with it is to replace with an editor / sed script the <> (or any other delimiter) with [].

closed 2024-02-22 by StefanSchippers