StefanSchippers/xschem #114: Conditional format statement (simulator)

rafmag, 2023-08-15:
Hello,
I was recently doing an upgrade and noticed that some of the basic symbols were updated.
[image]
This new `save` line is causing a conflict with my Xyce simulation setup. 
It's simple enough to remove, you don't need to change anything. I was just wondering if there was a way to make that line conditional on the simulator being used, so that it gets ignored if I choose Xyce as the simulator. I can make the change on my side, no need to update anything on the source side if it's an issue.

StefanSchippers, 2023-08-15:
If you set the simulator to 'Xyce batch' in Simulation->Configure simulators and tools the produced netlist will have these .save lines removed.

More in general if needed I can create alternate `format` attributes in symbols, like `ngspice_format` or `xyce_format`, with different netlisting rules. This is already implemented in xschem, I only need to add these attributes to some symbols.

Giving the command `xschem set format xyce_format` before netlisting will use the `xyce_format` attribute for symbol netlisting if found, otherwise revert to the fefault, 'format'.

rafmag, 2023-08-15:
Can I `set format xyce_format` in the xschemrc ?

rafmag, 2023-08-15:
So in the symbol, instead of `format` I can have both `ngspice_format` and `xyce_format` and then pick which one I want by `set format xyce_format` or `set format ngspice_format` ?

StefanSchippers, 2023-08-15:
Exactly. It is advisable to leave also `format` as a fallback for spice netlists just in case no `set format` is given.

rafmag, 2023-08-15:
Thanks, that's exactly what I was looking for!

closed 2023-08-15 by rafmag

StefanSchippers, 2023-08-15:
If you find problems while using this feel free to reopen.

StefanSchippers, 2023-08-16:
> Can I `set format xyce_format` in the xschemrc ?
the `format` attribute name to be used for netlisting is set via the `xschem` tcl command.
xschem creates an additional tcl command called `xschem` to access/get/set internal C data structures.
this command is not available in the `xschemrc` file since the schematic data structures (and the command itself) have not yet been created.
If you are sending commands to xschem via scripts then simply do something like this:
`xschem set format xyce_format; xschem netlist`
If you want you can call xschem this way:
`xschem --command 'xschem set format xyce_format'`

May be in a next commit I will add a 'postinit_commands' variable (or similar name) that may be defined in the xschemrc file, containing a set of tcl commands that will be executed after completing startup.

StefanSchippers, 2023-08-16:
In latest xschem commit you can now set in xschemrc:
`set postinit_commands {xschem set format xyce_format}`

The `postinit_commands` variable may contain tcl commands (separated as usual by ;) that will be executed by xschem after completing startup, after all data structures have been created.