StefanSchippers/xschem #115: Increase GUI elements in HiDPI displays

niuniuaba, 2023-08-23:
Hello,

The GUI elements (buttons, fonts, icons, etc.) are too small in my laptop display for me to read. Is it possible to increase the sizes?

Thanks in advance.

StefanSchippers, 2023-08-23:
there is a `tk_scaling` variable in the `xschemrc` file.
if you do:
`set tk_scaling 2.0`
all GUI elements will be bigger.
default value for tk_scaling is 1.0, so you can increase or decrease the size.

niuniuaba, 2023-08-23:
Hello Stefan,

Thank you for quick response. Unfortunately that doesn't work (more precisely, not fully work). I set tk_scaling to 2.0 and it seems only the preview in insert symbol widget changes but almost all others doesn't change at all. I attach some screenshots (left: scaling 1.0; right: scaling 2.0)

[image]

[image]

Best regards.

StefanSchippers, 2023-08-23:
This is strange. I get this. image shows side by side xschem , the second on the right has tk_scaling set to 2.0
[image]

StefanSchippers, 2023-08-23:
Only the toolbar is not resized, this can be done with some small modifications but only by integer numbers, so it is easy to double/triple but not for fractional quantities.

StefanSchippers, 2023-08-23:
remember to do `set tk_scaling 2.0` (or any other number)  in the xschemrc file, so it will be effective before xschem creates widgets.

niuniuaba, 2023-09-02:
Hi there,

I tested in two PCs one Ubuntu 22.04 another CentOS variant, all in used with Nvidia graphic drivers, GNOME desktop and X11 server (not wayland), and tk_scaling doesn't work in any of the two.
It's quite difficult to track it down so I just configure fonts in xschem.tcl and everything looks great. I don't use the menubar so much and I think we can close this issue.

Best wishes.

StefanSchippers, 2023-09-02:
Can you please test this small tk program?
put following text in a test.tcl file:
```
tk scaling 3.0
text .a
.a insert 1.0 "Hello, World!"
pack .a
```
then execute the tcl-tk program:
`wish test.tcl`

you will see a text widget with some text in it.
Try to change the first line (`tk scaling 3.0`)
and run again the program. 
Does text size change?

niuniuaba, 2023-09-02:
Hi Stefan,

Nope, the text size doesn't change no matter what I put tk scaling to.

StefanSchippers, 2023-09-02:
Ok  this is consistent with xschem on your systems. May be there is a
desktop environment setting that forbids gui elements scsling.
Another possibility is to set a higher DPI setting in the system
configuration. Every system has different ways to do so, so you need to do
some research.

StefanSchippers, 2023-09-02:
Please try also this:
```
xschem --tcl 'font configure TkDefaultFont -size 16; font configure TkMenuFont -size 16'
```
this will change the default font and menufont .

If this works you can set these two lines in your xschemrc file:

```
font configure TkDefaultFont -size 16
font configure TkMenuFont -size 16
```

Of course change the size to your specific needs.

niuniuaba, 2023-09-03:
Hi there,
The font configure in xschemrc works. Much more convenient than modifying xschem.tcl file. Thank you.

closed 2023-09-03 by niuniuaba