StefanSchippers/xschem #116: xschem immediately closes remote connection after executing command

rafmag, 2023-09-09:
Simple example:

`set sock [socket localhost 2021]` xschem is listening on port 2021
`puts $sock "xschem get schname"`
`flush $sock` xschem closes the connection here...
`gets $sock line`
`puts $line`

The moment I flush the buffer xschem correctly runs the command and I get the right output in the `line` variable.
but then xschem closes the connection immediately, not allowing me to send more commands without re-running the socket command.

Am I doing something wrong here?

StefanSchippers, 2023-09-09:
The close connection is done on purpose.
This allows to read the response until an EOF is found.
Some xschem CLI commands return multiple lines of text. 
Since TCP communication is asynchronous and may be slow / with high latency it becomes difficult to read the response up to the last line / character.
If connection is kept alive there is no way you can tell if you have read all response lines or some data is still traveling the wires. As a result if you do one more gets call you get a deadlock. Receiver is waiting for data, transmitter (xschem) is not sending anything but connection is open.
the read command can not be used at all since it will deadlock if there is no EOF condition.
(from tcl read manpage: "_The read command reads all of the data from channelId up to the end of the file_").
The following is a tcl implementation of a communication procedure:

```
# Send a command to a running xschem instance over tcp
# 'addr' specifies a host and a port number separated by space(s)
# host and port refer to the host xschem is running in and the port number it is listening to.
proc cmd {c {addr {localhost 2021}}} {
  set cnt 0
  # try some times until socket gets opened, otherwise give up and return error
  while {[catch {eval socket $addr} fd]} {
    after 20  ;# wait 20ms
    incr cnt
    if {$cnt >= 15 } {return $fd} ;# failed: return error message
  }
  puts "socket opened after $cnt tries"
  puts $fd $c ;# send command to xschem over tcp 
  flush $fd
  set r [read $fd] ;# read xschem's response
  close $fd ;# done: close socket
  return $r
}
```
In above cmd procedure the read instruction reads all the response up to EOF, so no matter how long is xschem response and how much time it takes, it will reliably read all data, without implementing complex mechanisms, like non-blocking I/O, polling and so on.
The above procedure assumes an xschem instance has been started. For this reason it tries some times to open a socket just in case the xschem process has been started but has not yet completed its own initialization.
When another command is sent this procedure will just open again the socket.
Example of the cmd usage:
```
set inst_list [cmd {xschem instance_list}]
```

If someone knows how to send an EOF condition without closing the TCP connection that would be great. I did not find anything better than just spitting out the response and closing the connection. 
This way the receiver should treat the socket as a regular file, no special read tricks are needed.

rafmag, 2023-09-09:
I was a bit concerned about the performance impact of your first approach, but it seems to run pretty quickly for you. I will implement that way and see how it goes. As always, thanks for the very quick and detailed response!

rafmag, 2023-09-10:
Your first approach worked really well, except for the several 100s of: 

`Close 127.0.0.1 46669`
`Accept sock5637c52d7e20 from 127.0.0.1 port 37417`

that get printed to screen, but I can live with that. Closing...

closed 2023-09-10 by rafmag

StefanSchippers, 2023-09-10:
I have removed all open/close messages and in/out data from console output. These will be printed only if xschem is started in tcl debug mode (`xschem -d -1`). It's now quite a while the tcp interface is in place so there is no more need to have all these messages.