StefanSchippers/xschem #117: Combine netlisting and edit_netlist in a single button

niuniuaba, 2023-09-11:
Hi there,

When I do simulations I always expect netlist editor pops up when netlisting finishes the netlist generation so I can take a look before the simulation.
I know I can invoke netlist editor from menu but I am wondering if it is possible to combine these two actions to the netlisting menubutton. In shell (bash, for example) there is something like netlisting && edit_netlist and I think there must be something familiar in Tcl script. This behavior could be annoying to others so I am asking if it's possible to do it on my side.

Best wishes

StefanSchippers, 2023-09-11:
There are two options. 
If you enable the option "`Show netlist win`"  in the Options menu a window will pop up with the netlist after netlisting. You can edit and save the file if you wish.
[image]

If you want to use the configured editor for xschem then you need to combine the following two commands:
xschem netlist
edit_netlist [xschem get netlist_name fallback]
You can do this by instantiating a launcher and defining this attribute in it:
`tclcommand="xschem netlist; edit_netlist [xschem get netlist_name fallback]"`

[image]

When you do a ctrl-left mouse button click on it a netlist is produced and editor is started:
[image]

StefanSchippers, 2023-09-11:
There is a third option: redefine the "`Netlist`" button command:
at the xschem prompt:
```
.menubar.netlist configure -command {xschem netlist -erc; edit_netlist [xschem get netlist_name fallback]}
```
If your xschem is recent you can add this in your xschemrc file:
```
###########################################################################
#### TCL COMMANDS TO BE EXECUTED AFTER COMPLETING STARTUP
###########################################################################
set postinit_commands {
   .menubar.netlist configure -command {xschem netlist -erc; edit_netlist [xschem get netlist_name fallback]}
}
```

niuniuaba, 2023-09-12:
Wow, what a quick response!
I prefer the last option because I use gvim as netlist editor and it's quite flexible to maintain a local xschemrc file.

Thank you and best wishes.

closed 2023-09-12 by niuniuaba