StefanSchippers/xschem #118: if i include an OPAMP model in a code-block its missing in the SPiCE-netlist

olisnr, 2023-09-20:
[3-state-dac-04.sch.zip](https://github.com/StefanSchippers/xschem/files/12676720/3-state-dac-04.sch.zip)

the netlist ends like this:

_...
+ Mje=.35786 Cjc=3.1745E-12 Vjc=.77077 Mjc=.19463 Tf=281.52E-12 Xtf=1.6571E3 Vtf=945.94 Itf=10.142 Tr=121.92E-9
+ Xtb=1.5000 Vceo=40 Icrating=0.2 mfg=Rohm)
.model UMT3906 PNP(Is=10.000E-15 Bf=214.10 Vaf=39 Ikf=.36736 Ise=6.2628E-12 Ne=1.9555 Br=14.872
+ Var=5.4000 Ikr=2.3066E-3 Isc=1.2518E-9 Nc=2.4667 Nk=.78931 Re=.3 Rb=2.1776 Rc=.38693 Cje=7.4001E-12 Vje=.73481
+ Mje=.32446 Cjc=5.4100E-12 Vjc=.55262 Mjc=.35478 Tf=352.19E-12 Xtf=4.0704 Vtf=1.2425E3 Itf=.18254 Tr=10.101E-9
+ Xtb=1.5000 Vceo=40 Icrating=0.2 mfg=Rohm)

*Rev-B, March-2010
*Modified notes
*
*//////////////////////////////////////////////////////////////////////
* (C) National Semiconductor, Corporation.
* Models developed and under copyright by:
* National Semiconductor, Corporation.
*/////////////////////////////////////////////////////////////////////
* Legal Notice:
* The model may be copied, and distributed without any modifications;
* however, reselling or licensing the material is illegal.
* We reserve the right to make changes to the model without prior notice.
* Pspice Models are provided AS

**** end user architecture code
**.ends
.GLOBAL GND
.GLOBAL VDD
.end_

the model is missing. it should look like:

_*Rev-B, March-2010
*Modified notes
*
*//////////////////////////////////////////////////////////////////////
* (C) National Semiconductor, Corporation.
* Models developed and under copyright by:
* National Semiconductor, Corporation.  
*/////////////////////////////////////////////////////////////////////
* Legal Notice:  
* The model may be copied, and distributed without any modifications;
* however, reselling or licensing the material is illegal.
* We reserve the right to make changes to the model without prior notice. 
* Pspice Models are provided "AS IS, WITH NO WARRANTY OF ANY KIND" 
*////////////////////////////////////////////////////////////////////
* For more information, and our latest models,
* please visit the models section of our website at
*       http://www.national.com/models/
*////////////////////////////////////////////////////////////////////
* BEGIN LMV981 MODEL
* MODEL TEMPERATURE RANGE IS -40 C TO +125 C, NOT ALL PARAMETERS TRACK THOSE 
* OF THE REAL PART VS TEMPERATURE
* FEATURES:
* OPEN LOOP GAIN AND PHASE, INPUT BIAS CURRENT, INPUT CAPACITANCE
* INPUT COMMON MODE VOLT RANGE, RAIL TO RAIL INPUT STAGE, WITH OFFSET TRANSITION
* INPUT CLAMPS TO RAILS,  CMRR WITH FREQUENCY EFFECTS, PSRR WITH FREQUENCY
* EFFECTS, SLEW RATE, QUIESCENT CURRENT, RAIL TO RAIL OUTPUT STAGE, HIGH CLOAD
* EFFECTS, CLASS AB BIAS IN OUTPUT STAGE, OUTPUT CURRENT THROUGH SUPPLIES
* OUTPUT CURRENT LIMITING, OUTPUT CLAMPS TO RAILS, OUTPUT SWING VS OUTPUT CURRENT
* SHUTDOWN, ENABLE TIME, QUIESCENT CURRENT DURING SHUTDOWN
*///////////////////////////////////////////////////////////////////////////
*
* PINOUT ORDER +IN -IN +V -V OUT NSD
* PINOUT ORDER  1   3   6  2  4   5
.SUBCKT LMV981 1 3 6 2 4 5
Q20 7 8 9 QLN
R3 10 11 2
R4 12 11 2
R10 8 13 1E3
R11 14 15 1E3
R12 16 6 4
R13 2 17 4
R16 18 19 300
R17 20 21 4
R18 9 22 4
D5 23 6 DD_

StefanSchippers, 2023-09-20:
Please send a zip with the text files of the models you want to include in the netlist, the top level schematic and also the `lmv981.sym` symbol xschem file.

StefanSchippers, 2023-09-20:
In your case the problem is that the model text is placed inside a value="
...
...
"
attribute

If inside the value quotes your text has quotes they must be escaped.
Exaample:

value="this is a text string with \"quotes\" inside"

If you look you have this text in the spice model:

`* Pspice Models are provided "AS IS, WITH NO WARRANTY OF ANY KIND" ....`

Change the text to:

`* Pspice Models are provided \"AS IS, WITH NO WARRANTY OF ANY KIND\" ....`

In xschem you can edit a code block, press 'q', set the attribute to value and then press 'Load' to load a file. In this case xschem will escape quotes (if any) for you.
[image]

olisnr, 2023-09-21:
great! thanks a lot!

closed 2023-09-21 by olisnr