StefanSchippers/xschem #119: Xschem under UBUNTU / MiNT has problems with screen refresh.

olisnr, 2023-09-20:
one example is:
i select an voltage source and deselect it after. after deselecting the circle is missing. 
[image]
[image]
[image]

there are also other problems at drawing lines or wires and at moving objects. while moving it produces a mess under the object and sometime it stays until one forces a redraw.

olisnr, 2023-09-20:
System:    Kernel: 5.15.0-84-generic x86_64 bits: 64 compiler: N/A Desktop: Cinnamon 5.2.7 
           wm: muffin dm: LightDM Distro: Linux Mint 20.3 Una base: Ubuntu 20.04 focal 
Machine:   Type: Laptop System: LENOVO product: 20S0000NMZ v: ThinkPad T14 Gen 1 serial: <filter> 
           Chassis: type: 10 serial: <filter> 
           Mobo: LENOVO model: 20S0000NMZ v: SDK0J40697 WIN serial: <filter> UEFI: LENOVO 
           v: N2XET39W (1.29 ) date: 07/25/2023 
Battery:   ID-1: BAT0 charge: 39.6 Wh condition: 44.8/50.5 Wh (89%) volts: 12.4/11.5 
           model: SMP 5B10W13906 serial: <filter> status: Discharging 
           Device-1: hidpp_battery_1 model: Logitech MX Ergo Multi-Device Trackball 
           serial: <filter> charge: 100% (should be ignored) status: Discharging 
CPU:       Topology: Quad Core model: Intel Core i7-10510U bits: 64 type: MT MCP arch: Kaby Lake 
           rev: C L2 cache: 8192 KiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 36799 
           Speed: 4300 MHz min/max: 400/4900 MHz Core speeds (MHz): 1: 4300 2: 4305 3: 4300 
           4: 4300 5: 4300 6: 4300 7: 4300 8: 4300 
Graphics:  Device-1: Intel UHD Graphics vendor: Lenovo driver: i915 v: kernel bus ID: 00:02.0 
           chip ID: 8086:9b41 
           Display: x11 server: X.Org 1.20.13 driver: modesetting unloaded: fbdev,vesa 
           resolution: 1920x1080~60Hz 
           OpenGL: renderer: Mesa Intel UHD Graphics (CML GT2) v: 4.6 Mesa 21.2.6 
           direct render: Yes 
Audio:     Device-1: Intel vendor: Lenovo driver: sof-audio-pci-intel-cnl bus ID: 00:1f.3 
           chip ID: 8086:02c8 
           Sound Server: ALSA v: k5.15.0-84-generic 
Network:   Device-1: Intel Wireless-AC 9462 driver: iwlwifi v: kernel port: 4000 bus ID: 00:14.3 
           chip ID: 8086:02f0 
           IF: wlp0s20f3 state: up mac: <filter> 
           Device-2: Intel Ethernet I219-V vendor: Lenovo driver: e1000e v: kernel port: efa0 
           bus ID: 00:1f.6 chip ID: 8086:0d4f 
           IF: enp0s31f6 state: down mac: <filter> 
           IF-ID-1: Z43VPN state: unknown speed: N/A duplex: N/A mac: N/A 
Drives:    Local Storage: total: 1.82 TiB used: 908.29 GiB (48.8%) 
           ID-1: /dev/nvme0n1 vendor: Kingston model: SNVS2000G size: 1.82 TiB speed: 31.6 Gb/s 
           lanes: 4 serial: <filter> 
Partition: ID-1: / size: 100.69 GiB used: 87.91 GiB (87.3%) fs: ext4 dev: /dev/nvme0n1p5 
USB:       Hub: 1-0:1 info: Full speed (or root) Hub ports: 12 rev: 2.0 chip ID: 1d6b:0002 
           Device-1: 1-1:2 info: Alcor Micro AU9540 Smartcard Reader type: Smart Card driver: N/A 
           rev: 2.0 chip ID: 058f:9540 
           Device-2: 1-8:3 info: IMC Networks Integrated Camera type: Video driver: uvcvideo 
           rev: 2.0 chip ID: 13d3:5406 
           Device-3: 1-10:4 info: Intel type: Bluetooth driver: btusb rev: 2.0 chip ID: 8087:0026 
           Hub: 2-0:1 info: Full speed (or root) Hub ports: 6 rev: 3.1 chip ID: 1d6b:0003 
           Hub: 3-0:1 info: Full speed (or root) Hub ports: 2 rev: 2.0 chip ID: 1d6b:0002 
           Hub: 4-0:1 info: Full speed (or root) Hub ports: 2 rev: 3.1 chip ID: 1d6b:0003 
Sensors:   System Temperatures: cpu: 39.0 C mobo: N/A 
           Fan Speeds (RPM): fan-1: 0 
Repos:     No active apt repos in: /etc/apt/sources.list 
           Active apt repos in: /etc/apt/sources.list.d/kicad-kicad-6_0-releases-focal.list 
           1: deb http: //ppa.launchpad.net/kicad/kicad-6.0-releases/ubuntu focal main
           Active apt repos in: /etc/apt/sources.list.d/oem-sutton.simon-bailee-meta.list 
           1: deb http: //lenovo.archive.canonical.com/ focal sutton
           2: deb http: //lenovo.archive.canonical.com/ focal sutton.simon
           Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list 
           1: deb https: //mirror.init7.net/linuxmint una main upstream import backport
           2: deb http: //archive.ubuntu.csg.uzh.ch/ubuntu focal main restricted universe multiverse
           3: deb http: //archive.ubuntu.csg.uzh.ch/ubuntu focal-updates main restricted universe multiverse
           4: deb http: //archive.ubuntu.csg.uzh.ch/ubuntu focal-backports main restricted universe multiverse
           5: deb http: //security.ubuntu.com/ubuntu/ focal-security main restricted universe multiverse
           6: deb http: //archive.canonical.com/ubuntu/ focal partner
           Active apt repos in: /etc/apt/sources.list.d/picoscope7.list 
           1: deb https: //labs.picotech.com/picoscope7/debian/ picoscope main
           Active apt repos in: /etc/apt/sources.list.d/qucs-qucs-focal.list 
           1: deb http: //ppa.launchpad.net/qucs/qucs/ubuntu focal main
           No active apt repos in: /etc/apt/sources.list.d/spotify.list 
Info:      Processes: 308 Uptime: 1h 21m Memory: 46.71 GiB used: 3.38 GiB (7.2%) Init: systemd 
           v: 245 runlevel: 5 Compilers: gcc: 9.4.0 alt: 9 Client: Unknown python3.8 client 
           inxi: 3.0.38 

Xschem 3.4.1

StefanSchippers, 2023-09-20:
Try this.(man xorg.conf)  in your /etc/X11/xorg.conf (or /etc/X11/xorg..somewhere) :

- Try to set Driver "Intel" instead of "Modesetting" in the "Device" section

- section set the PageFlip option to false, example (this gives to me a **HUGE** speed improvement in 2D drawing).
- 
```
...
...
Section "Device"
     Identifier  "intel"
     Driver      "intel"
     Option "PageFlip" "false"
 EndSection
...
...
```
then restart the system and test.

If nothing changes disable Acceleration altogether (of course just for testing):

```
 
...
...
Section "Device"
     Identifier  "intel"
     Driver      "intel"
     Option "Accel" "false
 EndSection
...
...
```

If by disabling acceleration things work OK then the driver implementation is broken.
Usually there are some options (man intel) to make things work. 

Another test is to start a VNC server and use xschem from within a vnc viewer session. VNC has a clean and compliant implementation of the X11 display server.

Your display graphic card is probably lot newer than mine, so may be I am missing some other option.

olisnr, 2023-09-20:
thanks a lot!

the Option "PageFlip" "false" is working for xschem. but now crashes the desktop environment cinnamon... do You think its possible to desable this option for xschem only?

StefanSchippers, 2023-09-20:
No, unfortunately I don't think a per application configuration is possible.
Did you restart the system completely? , ( I mean not only logging out of the session)...

olisnr, 2023-09-20:
yes i rebooted.

but i found out, that the problem for cinnamon seems to be the Driver "intel", i think it should be "i915" or  "i915 v". using "i915" cinneman dont crashes but xschem has the same problems. 

i have to tray othere settings, like the "Accel" "false"...

StefanSchippers, 2023-09-20:
You should leave "Modesetting" and then set the PageFlip option.
I think modern desktop environments want to use Modesetting.

olisnr, 2023-09-20:
/etc/X11/xorg.conf : 

Section "Device"
    Identifier "intel"
    Driver "Modesetting"
    Option "PageFlip" "false"
    Option "Accel" "false"
EndSection

dont work for my Xschem.

is there some ./configure option i could try?

StefanSchippers, 2023-09-20:
No, there is no configure option, the problem is in the Xserver incorrectly doing some 2D operations. But I think playig with the various xorg.conf options there will be a combination that just works.

olisnr, 2023-09-20:
the strange thing is, all other 2D things i need are working: inkscape, kicad, libreOffice, microchips libero and even ltspice over wine. 
can it bee that Xschem is using a graphic library that i should compile my self?

StefanSchippers, 2023-09-20:
No, xschem is using direct Xlib calls, it only uses Cairo for antialiased text. 
There is no middleware library for graphics.

I will eventually switch to cairo graphics some day, but as with any middleware lib there will be a 5x speed penalty to pay.

For the modesetting driver (man modesetting) the acceleration is disabled by:
 `Option "AccelMethod" "none"`
can you please try that also?

if drawing is correct with no acceleration and broken with acceleration there is really some missing drawing  primitive in the accelerated driver.
Most programs nowadays use openGL or some higher level libraries.
xschem uses old school 2D graphics, it was developed first on SunOS, then Solaris, then Linux. I have ported the program on numerous generations of PCs, from Pentium2/64MB RAM to Corei7/many GB RAM.

I also have machines with AMD radeon GPUs where some options need to be set in the xorg.conf to avoid garbage in the xschem display.

Usually when you disable acceleration software rendering is used. The Xserver core uses low level software drawing primitives written by Keith Packard some 45 years ago, and for pure 2D these are often faster than any accelerated GPU. GPUs are optimized for 3D workloads, there is no benefit at all for 2D. 

But most importantly Keith Packard implementation of the graphic stack is guaranteed to produce correct results down to the single pixels. The specification of all drawing shapes, raster operations, bit blitting etc. does not leave any space for interpretation or ambiguity.

If software rendering is OK and accelerated rendering is broken there is **very** likely a problem in the driver.

Again, I believe there is some combination of options that will work for you, without disabling acceleration, however there is no one recipe that fits all gpu models.

olisnr, 2023-09-20:
Section "Device"
    Identifier "intel"
    Driver "Modesetting"
    Option "PageFlip" "false"
    Option "AccelMethod" "none"
EndSection

is working!!!

and now i should test all combinations of the dirver "Modesetting"? what is a good starting point?

Option "SWcursor" "boolean"
Option "kmsdev" "string"
Option "ShadowFB" "boolean"
Option "DoubleShadow" "boolean"
Option "AccelMethod" "string"
Option "PageFlip" "boolean"
Option "VariableRefresh" "boolean"
Option "AsyncFlipSecondaries" "boolean"
Option "ZaphodHeads" "string"
Option "ZaphodHeads" "LVDS,VGA-0" 
Option "UseGammaLUT" "boolean"

shadowFB? may be?

StefanSchippers, 2023-09-20:
Yes you should test with acceleration enabled and various combinations of PageFlip and ShadowFB.  The other options I really don't know what they do , so i can not help much.

StefanSchippers, 2023-09-21:
Just some explanations of how xschem handles selection / redrawing of objects.

The main problem is that if you select some objects xschem draws them in grey. When unselecting the grey areas must be restored. There are many approaches to do this.

1. redraw the whole screen when user unselects objects. This very inefficient, expecially is user selects only a small amount of the whole circuit.
2. Redraw only the screen rectangle that contains the selection. This is better, than 1. however if user selects one object in the lower right corner and one in the upper left corner you are back in the same situation of point 1.
3. Redraw many screen rectangles each containing one selected object. 
4. Redraw only the selected objects using an offscreen save buffer as the fill pattern. This requires redrawing **only** the selected objects instead of entire portions of the screen. This is the approach used by xschem and explained below.

Xschem uses a double buffering technique. Drawing is done in an offscreen pixmap (think of it as a canvas that is not displayed in any window, it sits just in main /GPU memory)
Image below shows the situation. xschem window with some objects selected (left) and a representation of the offscreen pixmap (right). when image in the pixmap is finished it is copied in the main window. If the main window gets obscured and then brought back to front xschem restores the content by copying the pixmap (or a portion)  to it

[image]

Now, if user clicks an empty area in the window, objects get unselected, xschem redraws the grey objects using the pixmap on the right as the color for every pixel.
This way xschem only needs to draw 4 rectangles, one circle and 3 lines and the image is restored. Note that if there are overlapping objects (as in the example image) you can not redraw the objects with their normal color, as this will erase parts of the overlapped objects.

This technique of using a secondary pixmap as the fill pattern is standard in X11, and for vector graphics it is often more efficient than blitting a big rectanble area, expecially on UHD graphics. If you have a 1920x1080 fullscreen xschem window and select a diagonal line from (0,0) to (1920, 1080) when unselecting xschem redraws just one line with the save pixmap as fill pattern for every line pixel. Game over.

Restoring by bit blitting requires to draw the whole screen (blit areas must be rectangular) so some MB of memory must be moved either in the GPU or in main RAM. In either case drawing a ~2200 pixel line using the secondary pixmap as the "pen" is often faster than copying ~8MB of video memory (in 32 bit /pixel a full HD screen requires ~ 8MB) even with acelerated GPUs.

The problem with some modern video cards is that the drawing operations using pixmap as draw/fill patterns are often not implemented correctly.
What I plan to do is to use the above option 3. as an alternate drawing method.
This will however take some time.

StefanSchippers, 2023-09-21:
As another example, when you select objects and press the 'm' (move) key, every time you move the mouse xschem does:

1. draw the selected objects using the secondary pixmap as fill/color for every grey pixel shown. This will clear the selection and restore the drawing (in the example the baclk background and the grey grid points are restored.
2. move the objects according to the mouse position
3. draw the grey selection to screen.
the save pixmap is not changed and contains the xschem drawing. The pixmap is regenerated when the move operation is finished by the user. It will be used to restore window in following operations.

[image]

olisnr, 2023-09-21:
what is the reason for the different layers whit the different colors? is this also used for optimization? 
would it be a bad idea to make an option that forces a redraw if an editing (move or deselect) is finished (as simple fix)?
is there an easy way to draw an xor cross at the cursor-tip (i would like this very much)?
the auto-join and trim function is working super, but the function to break a wire, i can use only at horizontal & vertical wires. is there a possibility to increase the pointing-tolerance of this function (to be able to break diagonal wires) ?
thanks a lot!

olisnr, 2023-09-21:
what do You think about nim-lang?

olisnr, 2023-09-21:
im not interested in the XOR, only in the big + cross! a fixed color is even better then XOR.

the braking is working on grid-points works good, but i need it mostly, when the line dont cross any grid-point. in this case the easiest way i found is to destroy the line and draw it new. but then i loos the connection, and thats a source of errors.

olisnr, 2023-09-21:
is it possible to find the next point of a line to the cursor? and brake it at the next grid-point?

StefanSchippers, 2023-09-21:
Of course it is possible, but the point on the grid can be very far away You can have oblique lines where only the start and end points are on grid. I think it is better to break the wire at the closest to cursor point and then if user wants align this point in the nearest grid.

olisnr, 2023-09-21:
sounds perfect.

to the question about refresh after an action. do You think this would be a total insult of the idea of resource saving software? or would it be simple to do it my self via TCL?

StefanSchippers, 2023-09-21:
It is not an insult. 

I have added the following line in `src/xschem.h`:

`#define FIX_BROKEN_TILED_FILL 1`

It is commented normally.

Update your xschem sources, uncomment the above mentioned line, rebuild the program
and please do a test. I can not guarantee anything because I don't have these graphic artifacts on my GPU.  This fix is also normally done on the windows version since on Windows there is no XsetTile function at all so tiled fill drawing does not work at all.

I have run the drawing tests with the above #define uncommented and performance impact is negligible, btw.

StefanSchippers, 2023-09-22:
Current xschem also handles cutting oblique lines. cut point is moved on closest snap

Uploading output.mp4…

 point and broken wires are changed accordingly.

StefanSchippers, 2023-09-22:
I have now also added a different command (Alt-Shift-Right mouse button) to cut wires without aligning the cut point.

For your convenience consider there is already a wiring command that starts a new wire attaching it to the closest start-end point (Shift-W) 
This way you can always connect a wire to another wire endpoint that is not on grid.

StefanSchippers, 2023-09-22:
Crosshair at mouse coordinates is also implemented now. It will show if:

`set draw_crosshair 1`

is set in xschemrc

There are probably some corner cases to fix. If you find any please report.

[image]

olisnr, 2023-09-22:
You are an hero! 

the breaking off-grid is working great. 

there are some little things
1. there is no ~/.xschem/xschemrc after "make install" nor after running Xschem. 
2. the i dont like the alt key, because its only on the left side. commands like rotate or flip should have an easier access, like cntr as example. would You consider to change the standard-key bindings? (also undo and redo is mostly cntr-z & y)
3. i would like to get rid of the cntr-s save warning-dialog. would it be ok to make this an xschemrc option?

then i dint found the meaning of the Fg & Status checking-box in the simulationtool dialog help. what is there meaning?
[image]

the driver fix definition i have to test in the evening on a other computer.

thanks a lot!!!

StefanSchippers, 2023-09-22:
The fg and status meanings are documented in the Help window that pops up if you press Help:

**Foreground (Fg) checkbutton tells xschem to wait for child process to finish.**

**Status checkbutton tells xschem to report a status dialog (stdout, stderr,
exit status) when process finishes**

if a subprocess is launched in foreground xschem will stop and be not responding until process is finished. This usually good for tasks that end immediately.
For simulations and other tasks that take time dont check `fg`
the `status` checkbox will pop up a dialog box when the process ends with some information (the process output , errors if any and exit status (Completed or Failed)

StefanSchippers, 2023-09-22:
For the Alt key issue, on my keyboard I have also a Alt Gr key on the right, that works the same as Alt. Can you please test?

StefanSchippers, 2023-09-22:
For the `xschemrc`, there is a system `xschemrc` placed after `make install` in:

` .../share/xschem/xschemrc`

you should copy this template into `~/.xschem/`

It is not mandatory to have `~/.xschem/xschemrc`, if no such file is found xschem starts with a default configuration which allows you to see all example circuits.

olisnr, 2023-09-22:
yes You are right. i didnt read the help-text correct. sorry for that! 

rotation works only on "Alt" on the left side and not for "AltGr" on the right. flip is the same. i use an "swiss-german" keyboard layout.

StefanSchippers, 2023-09-22:
You are right, after switching keyboard layout to 'de' (german) the right alt key (AltGr) does not work.

Will investigate

olisnr, 2023-09-22:
would it be a god idea to copy the default xschem to the ~/.xschem directory, if there isnt allredy one? it would then have the same logic like the simrc file. delete it for reseting the stettings. the default xscemrc is great, because of the comments. its the configuration and documentation at the same time.

olisnr, 2023-09-22:
where is the U for undo coming from? is this standard in some software?

StefanSchippers, 2023-09-22:
Yes it is a common keybinding in Cadence Virtuoso schematic editors.

StefanSchippers, 2023-09-22:
Ok I understood the Alt-Gr issue. 

On most non US keyboards the Alt-Gr key is used to generate extended characters (often requiring 2 bytes instead of a single byte ASCII code). 

So `Alt-Gr` + `r` does not generate the 'r' character but some strange characters, like: ¶

Since this mapping is different for all foreign keyboards I think I can not use this key, it is a nightmare to accommodate all keyboard layouts.

On US keyboards there are no such extended characters so the Alt-Gr works just like an additional Alt key.

I have modified xschem to accept the windows keys as alternatives for Alt. Some keyboards (but not all) have this key on both sides.

I can not do a massive change of default keybindings since there are many users now.
It is possible to redefine keybindings locally in xschemrc

StefanSchippers, 2023-09-22:
for example these two lines in xschemrc:
```
set replace_key(Control-z) u        ;# undo
set replace_key(Control-y) Shift-U  ;# redo
```
will set `ctrl-z` and `ctrl-y` for undo/redo

You will lose the existing ctrl-z key (zoom out) you can use the 'u' key for that:

`set replace_key(u) Control-z         ;# zoom out`

so you swap `ctrl-z` and `u`

olisnr, 2023-09-22:
yes i will change the key-bindings a little bit. thanks for the examples.

what do You think about removing the safty dialog at cntr-s? and the writing a default xschemrc like the simrc?

StefanSchippers, 2023-09-22:
I will add an option for removing the ctrl-s warning, 
for placing a default xschemrc it can be done if there is no already one there (i don't want to overwrite user settings). Will work on that

olisnr, 2023-09-22:
great!

i tried the new nice cross cursor, and it seems to make some problems if i want to insert some parts:

[image]
[image]
[image]
[image]

olisnr, 2023-09-22:
if the FiX is commented, then my screen looks like this after moving the mouse:

[image]

StefanSchippers, 2023-09-22:
Yes, this is the reason for the FIX_BROKEN_TILED_FILL

olisnr, 2023-09-22:
safe. now i tested the cross also on a NViDiA driver. and that worked perfect without the fix.

StefanSchippers, 2023-09-22:
I think the crosshair (with the FIX_BROKEN_TILED_FILL) is now improved. Can you please test again the reentrant bbox error?

olisnr, 2023-09-22:
inserting and moving are working now. 

its really nice to use now!

what do You think about selecting via drag a rectangle via mouse. some CADs does select different if You drag from left to right (only the things inside the rectangle), then from right to left (all things that are touched)? would this be usefull in xschem, or better not?

StefanSchippers, 2023-09-22:
yes it is interesting. However i have added many things lately. My plan is now:
- do some testing and stabilize bugs (your help is welcome)
- when all things are stable i want to tag a version 3.4.4 (it is now 3.4.4 release candidate) 

New features will eb added after that.

olisnr, 2023-09-22:
super! thanks a lot!!!

StefanSchippers, 2023-09-23:
latest xschem has an xschemrc tcl variable:
`set no_ask_save 1` 
by default it is set to 0 of course.
You get rid to the ask save dialog boxes when saving modified schematics.
Be careful!

rafmag, 2023-09-23:
> what do You think about selecting via drag a rectangle via mouse. some CADs does select different if You drag from left to right (only the things inside the rectangle), then from right to left (all things that are touched)? would this be usefull in xschem, or better not?

I upvote this idea too (when you have time to work on it of course) I have been missing this ability, but didn't think of bringing it up.

StefanSchippers, 2023-09-23:
I Will implement this after 3.4.4 release, then a new dev. cycle begins

I need to know what happens is user starts in one direction an then changes
idea and drags in the other direction.

olisnr, 2023-09-23:
i would decide from the sign of the distance of the current cursor, to the cursor-position of the start. sot that if You change the direction and go over the starting point the selection type changes

rafmag, 2023-09-23:
> I need to know what happens is user starts in one direction an then changes idea and drags in the other direction.

You go by: sign(start click X coord - release click X coord)

StefanSchippers, 2023-09-24:
> would it be a god idea to copy the default xschem to the ~/.xschem directory, if there isnt allredy one? it would then have the same logic like the simrc file. delete it for reseting the stettings. [...]

@olisnr 
I have added this function. On startup if there is no `$USER_CONF_DIR/.xschem` (usually `~/.xschem`) it is created. If this happens I copy now also the template `xschemrc` into it.

StefanSchippers, 2023-09-26:
The compile option FIX_BROKEN_TILED_FILL is now also a runtime option.
View menu -> "Fix for GPUs with broken tiled fill"
so you don't need to rebuild the program to enable the graphic workaround.

olisnr, 2023-10-11:
after an update to the next linux-mint version (based on ubuntu 22.04), the driver-problems are gone...

closed 2023-10-11 by olisnr