StefanSchippers/xschem #121: question about ZOOMing

olisnr, 2023-09-23:
if i zoom out and then zoom in, without moving the mouse in between, i dont come to the same section back. is this intended or some problem?

[image]
[image]
[image]

StefanSchippers, 2023-09-23:
Yes this is intended. However behaviour can be changed in xschemrc: 
uncomment and set following variable to 1:

#### if set to 0, when zooming out allow the viewport do drift toward 
the mouse position,
#### allowing to move away by zooming / unzooming with mouse wheel
#### default setting: 0
# set unzoom_nodrift 0

olisnr, 2023-09-23:
Xschem is Xtrem: most ideas are already implemented!

thanks a lot!

closed 2023-09-23 by olisnr