StefanSchippers/xschem #122: most of the times i use cntr-mouse-drag to select

olisnr, 2023-09-23:
is there a possibility to swap the cntr-mouse-drag select with the mouse-drag? i searched in the xschemrc but dident found anything.

if not, i can try to do this. the line in select.c:1314 cold make an XOR whit a variable defined in xinit.h and set in options.c. would be this the way?

StefanSchippers, 2023-09-23:
I have done the change in callback.s, in these lines:

```
     BEFORE                                         AFTER
===============================================================================================
     if(state & ControlMask) {                     if(state & ControlMask) {
       tclsetvar("enable_stretch", "1");             int es = tclgetboolvar("enable_stretch");
                                                     tclsetboolvar("enable_stretch", !es);
       select_rect(END,-1);                          select_rect(END,-1);
       tclsetvar("enable_stretch", "0");             tclsetboolvar("enable_stretch", es);
       break;                                        break;
     } else {                                      } else {

```

so control-drag will flip `enable_stretch` instead of setting it to 1.
The initial state can be already set with `Options->Enable stretch`
This way control-drag and drag can be swapped. Thank you for the hint.

olisnr, 2023-09-24:
thats great. thanks!

olisnr, 2023-09-25:
wouldnt it be good to also implement this logic to the cntr-m vs m command? because if im in the drag-wires-mode, then i have the connections saved when i move a rectangle selected part, but not a per click selected part.

if the cntr-m would be standard for the "enable stretch" mode, then both selection types would save the wireing. what do You think?

StefanSchippers, 2023-09-25:
Yes this makes very sense. Will do that soon.

olisnr, 2023-09-25:
wouldnt be the "auto join/trim enable" be better in the option menue? near the "enable stretch"?

olisnr, 2023-09-25:
what does the tools "remember last command"?

StefanSchippers, 2023-09-25:
Please update xschem

- the m and ctrl-m commands are swapped if `enable stretch` is checked.
- `Remember last command` menu entry renamed to `Persistet wire/line place command`. see video clip for explanation.
- Auto join/trim wires moved to Options menu.

https://github.com/StefanSchippers/xschem/assets/69359491/f0d9a2c0-6b96-48d7-9c30-bfb3bf37e073

StefanSchippers, 2023-09-25:
I can move it to options, but something must get out of options as it gets too long

the two variables for xschemrc are:  `persistent_command` and `enable_stretch`

StefanSchippers, 2023-09-25:
ok, done:

"Persistet wire/line place command" goes to Options, 
"Show net names on symbol pins" goes to Symbol, 
"Show netlist after netlist command" goes to Simulation

olisnr, 2023-09-25:
i think there is something broken now:

selecting
[image]
after pressing "m" a 2nd pice of the wire is also selected
[image]
after moving the wire is broken
[image]

StefanSchippers, 2023-09-25:
Yes, i have reproduced the issue. will investigate asap

StefanSchippers, 2023-09-25:
Can you please update & test again?

olisnr, 2023-09-25:
there is a difference in moving if i u
se the cntr-m between clk selection and drag selection:

clk sel :: m
[image]

clk sel :: cntr-m
[image]

drag sel :: m
[image]

drag sel :: ctrl-m
[image]

olisnr, 2023-09-25:
but im not sure if this intended, because, in the case of click i dont select the points of the wire, and in the drag selection i do...

olisnr, 2023-09-25:
i think in the cntr-m situation, partial selections should be ignored.

StefanSchippers, 2023-09-25:
yes this is expected. You can always do a ctrl-drag to avoid selecting wire endpoints

olisnr, 2023-09-25:
ok. then there is only this selection expanding left of the "m" command. 

as example You select a part only, using click or cntr-drag, and then You want to use the non-dragging move command, but You press by mistake only the m, then the selection is expanded, and if You then press cntr-m You get the dragging movement.

StefanSchippers, 2023-09-25:
If you press the stretch move command (m in your case) some wire terminals are added to current selection before starting the move operation.

You can do an Alt-Click to unselect the partially selected wires.
May be another useful command to add is  "clear all partial selections".

olisnr, 2023-09-25:
wouldnt it be nicer for the user, if his selection isnt changed? i mean, You do all the things necessary for the movement, and if the command is completed You restore the users selection?

StefanSchippers, 2023-09-25:
Sorry, please be very precise, i don't understand the question. What are the actions you are thinking of and what is the behavior you expect?

olisnr, 2023-09-25:
if i select only a part and do a movement (dragging or only moving), after the movement i want to have the same part selected. 
while moving the selection can change or not, but if the movement is done, the same part should be selected like in the beginning.

StefanSchippers, 2023-09-25:
I don't understand.

Condider this, first

A move command will just move selected objects (_the selection_)
Full selected wires and symbols are moved as a whole.
Partially selected wires will get only the selected endpoint moved.
After the move operation is done all the selection remains the same.

The special "stretch move" command (m in your case) will just add some partial selects on wires that touch selected objects.
after completing the move operation all selections (full or partial) remain unchanged.

Clearing part of the selection before doing a "non stretch move" operation and restoring afterward is a non trivial change
Modifying the move operation to ignore partial selects is also non trivial.

Also in the default configuration:
doing a control-drag does a selection with partial selection of wires that are not completely inside the selection rectangle.
a following m command moves the seleced set, so partially selected wires will be stretched.
The above command sequences can not be changed since there are too many users that are now used to this.

Also moving some objects away without stretching the attached nets and leaving these partially selected nets selected after the move is a very dangerous situation, see image.
[image]

olisnr, 2023-09-25:
its about "The special "stretch move" command (m in your case) **will just add some partial selects** on wires that touch selected objects". 

would it be possible that the "stretch move" saves a copy of the actual selection, the add the partial wires, let the user move the selected around, and after ending the movement or after abort the command, restore the in the beginning copied selection? 

i think You do a part of this anyway. because, You do make a copy of the original part-positions, because You need it for the case if the user aborts the command.

olisnr, 2023-09-25:
like this:

select r
[image]

move r
[image]

place r
[image]

StefanSchippers, 2023-09-26:
I have added the option to unselect these partial wires. The tcl variable is: `unselect_partial_sel_wires` (default setting: not enabled (0) ). If set to 1 it does what you asked, if I understood well. This needs some testing.
There is also an entry in the Options menu that toggles this variable.
Thank you!

https://github.com/StefanSchippers/xschem/assets/69359491/aa27c8c9-9ad0-43a8-be26-6c840fb8ff18

olisnr, 2023-09-26:
thanks a lot that what i asked for, im happy to test it!

i was trying to find out how to do it my self, but Your work is big and complex and i dont understand how its woks. is there a description how the SW works internally?

closed 2023-09-26 by olisnr