StefanSchippers/xschem #123: T connections arent handled the same like PiNs

olisnr, 2023-09-24:
i think the T connections should be safed at moving using the stretch enabled. but in the moment Ts are broken. would it be possible to handle the Ts like the PiNs?

[image]
[image]
[image]
[image]

StefanSchippers, 2023-09-24:
See this short clip.

https://github.com/StefanSchippers/xschem/assets/69359491/47b62c42-6dc5-46cd-8c80-58c1b9b387b3

There is also another move command, in addition to stretch mode move, Shift-M, this will add wire segments to selected instance pins and wire endpoints, instead of stretching existing attached wires. You might want to test this one as well.

See the difference below

[image] 

[image]

olisnr, 2023-09-25:
great, i only need to "set autotrim_wires 1" in xschemrc. 

thanks!

closed 2023-09-25 by olisnr