StefanSchippers/xschem #125: with cross-hair active the "... : UNSAVED data ..." warning dialog-box hides the cursor

olisnr, 2023-09-29:
the open dialog and others show the cross-hair also in the drawing window, only the "... : UNSAVED data ..." warning dialog-box does it. 
may be the redrawing is blocked.

olisnr, 2023-09-29:
super

closed 2023-09-29 by olisnr

StefanSchippers, 2023-09-29:
An easy workaround is to set the mouse pointer to default whenever mouse pointer exits the drawing window. See video.

https://github.com/StefanSchippers/xschem/assets/69359491/dff5a2f3-0a1e-4e00-9103-73677469dd90

StefanSchippers, 2023-09-29:
Side note: you can set the layer (color) used for the Xhair with: 
**set crosshair_layer <n>** where n is a number. See xschem Layers menu for the layer numbers.
If an invalid number is given the grey layer (selection) is used.

olisnr, 2023-09-30:
yes its perfect! 

in You video, i saw, thats possible to send the "remote commands" direct via the terminal. are this the commands that are described in the comments of the xschemrc?