StefanSchippers/xschem #126: question: plot different simulations in one sheet?

olisnr, 2023-10-05:
i would like to plot different simulations in the same sheet, like in QUCS.
to setup different simulation seems easy via the TCL-commands, but i dont see a way how to bind the .raw-files to a plot-window.
is this possible in the moment?

StefanSchippers, 2023-10-05:
No this is not possible at the moment. You can load different raw files in
additional xschem tabs.
I want to enable different raw file for each waveform, so you can show the
same signal from different raw files in the same graph. This will be done
after 3.4.4 release.

StefanSchippers, 2023-10-14:
A first improvement is now done. Xschem already allowed to read multiple simulation runs from the same raw file, for example a bunch of transient simulations each with a different VCC value. However choosing the run to display in the graph was not flexible. Either display all runs overlayed for all signals in the graph or display a specified run for all signals in the graph. Now each signal can optionally indicate a specific run with the syntax:
`node%n` where n is an integer (starting from 0) indicating which run to display for node.
This allow s for example to have different runs of the same node in the same graph, like:
`node%2 node%4 node%13`.
See an example here.

In general you can write a string in the form:
`"alias; node % n"` so you will display a better description of the node (instead of node % n) 

[image]

I think a next improvement that has yet to be done is to associate a different raw file to a graph, so you can have more graphs each referring to a different raw file (and each raw file as described here may contain multiple runs of the same simulation type).

olisnr, 2023-10-14:
yes, thats also very nice. when i did some temp-sweeps normally i had to mark the lowest temperature with different step-spacing, like -40 -41 -20 0 20 40 60 80 100 120...

StefanSchippers, 2023-10-16:
Here is an example with raw file loading pushed to the limit.  4 raw files loaded, OP (values annotated in schematic), tran, dc, ac.
Each analysis done on 3 different bias currents.

Each graph may now specify a different raw file. Each raw file may contain multiple runs (of the same analysis).
Need to write some documentation and do a lot of testing.

[image]

olisnr, 2023-10-16:
You are a hero! i will never use LTspice again!

closed 2023-10-16 by olisnr