StefanSchippers/xschem #127: flipping multiselection

olisnr, 2023-10-06:
if i select more then one thing and use `alt-f` every thing flips for it self.

is it possible to flip it as group? (and is it possible to flip vertical instead of horizontal or do i need combinations of flipping and rotating?)
[image]
[image]
[image]

StefanSchippers, 2023-10-06:
Yes, the commands for flipping  and rotating as a whole the selection are Shift-F and Shift-R respectively.
They are documented in the Help-> Keys menu, but are missing in the Edit menu. WIll add these.
A vertical flip is easy to do as 2 rotations + 1 flip in any order.

After 3.4.4 i will also add a vertical flip command.
See short clip.

https://github.com/StefanSchippers/xschem/assets/69359491/e77e5ecd-e8a6-4c28-9d78-291791701c93

olisnr, 2023-10-06:
great. thanks a lot!

closed 2023-10-06 by olisnr