StefanSchippers/xschem #128: why isnt $netlist_dir/$n.raw working?

olisnr, 2023-10-06:
i try to find the TCL variable name for the current filename (without the ".sch"). to use in:

`tclcommand="xschem raw_read $netlist_dir/$n.raw"`

i tried $s $n $N $currname $node and others i found in `xschem.tcl`... but at the time i need it, the TCL command window says: `can't read "n": no such variable`

StefanSchippers, 2023-10-06:
The schematic name is in the C code, but there are tcl access functions.

set myname [xschem get schname]

Gets the full schematic name with all the path and ,sch extension.

set myname [xschem get current_name]

Returns only the schematic name, with .sch.

You can remove the extension with the icl file command;

set myname [file root $myname]

(If I remember well...)

StefanSchippers, 2023-10-06:
The schematic name is in the C code, but there are tcl access functions.
`set myname [xschem get schname]`
Gets the full schematic name with all the path and .sch extension.
`set myname [xschem get current_name]`
Returns only the schematic name, with .sch.
You can remove the extension with the icl file command;
`set myname [file rootname $myname]`

olisnr, 2023-10-06:
thanks. then i change the load command to:

```
name=h5
descr="load waves" 
tclcommand="
set doc [file rootname [xschem get current_name]];
xschem raw_read $netlist_dir/$doc.raw;
"
```

closed 2023-10-06 by olisnr