StefanSchippers/xschem #129: xschem dosent work from my terminal if i type "xschem &\n" but "xschem\n" works

olisnr, 2023-10-18:
if i use the `xschem &` from bash in a gnome-terminal, then the menus are blocked and the xschem window isnt refreshed if i change the window size. but xschem is woking itself

olisnr, 2023-10-18:
v3.4.5

olisnr, 2023-10-18:
`end process` dont work, only `kill process` to end the `&` started program

StefanSchippers, 2023-10-18:
This expected. Xschem uses the shell as a terminal for user interaction. If you want to launch xschem in the background you must detach it from stdin/stdout/stderr. Background processes can not communicatewith the terminal and if they attempt to poll they will block.

Use the -b option:

xschem -b &
should work fine.

olisnr, 2023-10-19:
could i try to write noblocking to stdout and stderr, and look if there is an timout or error, and then decide if it has tho use the -b mode? or wouldent that work because of some other things?

StefanSchippers, 2023-10-19:
Don't know exactly, but if you try to run the wish tk shell in background you have the same problem.
```
schippes@asus:~$ wish &
[1] 18276
schippes@asus:~$ 

[1]+  Stopped                 wish

```

StefanSchippers, 2023-10-19:
even the bash shell should not be run in background: it can't read/write to the terminal so it blocks may be trying to write the prompt:
```
schippes@asus:~$ sh &
[3] 18770
schippes@asus:~$ 

[3]+  Stopped                 sh
```

StefanSchippers, 2023-10-19:
command shells shoud not be run in the background, it is a non sense since their primary function is to interact via the terminal.
Xschem is a GUI app so can work without  CLI . This is the reason for the -b option. Turn off I/O from/to the terminal.

olisnr, 2023-10-20:
the thing is, that i never saw a frozen window in a GUi-SW after starting it form commandline and &... even magic (a really strane SW dont freezes).

im certain there is a way to fix it, but its more difficult then i thought.

StefanSchippers, 2023-10-21:
I think it is now fixed. At the very beginning (`main.c`) xschem checks its running process grpup ID (PGID) and if it is different from the controling terminal one then it self-detaches from the console (does not accept commands and does not print info there).
```
  /* if invoked in background detach from console */
  if(getpgrp() != tcgetpgrp(STDOUT_FILENO)) {
    cli_opt_detach = 1;
  }
```
Please test, if it works close the issue.

olisnr, 2023-10-21:
it works. thanks

closed 2023-10-21 by olisnr