StefanSchippers/xschem #131: Tab features

martinjankoehler, 2023-10-19:
- tooltip for tab showing the full path
- tab context menu (right click) could offer some actions, i.e. ideas could be
   - close
   - move to separate window
   - open schematic / symbol counterpart
   - copy path to clipboard
   - (open directory in File Manager)

StefanSchippers, 2023-10-19:
the tooltip in tabs showing the full pathname is now done. Was very quick (1 line of code)

Moving a tab to separate window is a bit tricky. The xschem architecture is to have either tabbed interface or multi-window, not both at the same time. Could be done easily by spawning another xschem process. All other actions could be easy to do with a ctx menu.

StefanSchippers, 2023-10-25:
The feature has been added.  Doing some more testing. If you have a chance please test. I like this addition!

[image]

martinjankoehler, 2023-11-08:
Hey @StefanSchippers, sorry for the delay (was on vacation).

I think in general it works great, many thanks for adding this!

Just updated to the latest master, and did some testing...

## Testing report

### Open symbol / schematic

When switching from a symbol-less schematic (i.e. a testbench, which is never used as a sub-block), it 
1. first shows an error message that the file does not exist
2. then it opens an empty symbol not yet saved, but already at the correct path

#### Issue: Error dialog with nonexistent counterparts

- So the fact that it does (2.) is good IMO,  as it would allow the user to create and save the symbol right away.
- But the error message (1.) is a bit irritating, still on the upside it establishes awareness that the file did not yet exist
- Instead of (1.), and after doing (2.), alternatives would be 
   - show an info message explaining that it does not exist yet, but can be created now
   - show nothing at all, the user might have enough awareness given the empty symbol

#### Issue: Menu item does not change

Steps to reproduce:
- Perform steps of the above last issue
- In the empty symbol now shown, open the tab menu
- EXPECTED: menu item is called "Open schematic"
- OBTAINED: menu item is called "Open symbol", and clicking it gives me a warning that the file is already open
- Solution: the tab menu item should be updated to "Open schematic" in case of symbols, and "Open symbol" in case of schematic

#### Issue: Warning dialog in case of already open counterparts

Steps to reproduce:
- Open both sch and sym (i.e. using the tab menu or other means)
- The tab menu item is correct (in constrast to the last issue when counterparts do not exist yet)
- EXPECTED: the tab of the counterpart should be activated
- OBTAINED: I get an warning dialog that the file is already open, the app stays at the current file and does not switch to the desired counterpart

### Open Terminal commands

might have to do with my environment, but steps to reproduce:
- open schematic
- click "open circuit dir. term."
- EXPECTED: xterm shell is opened, and changes the current working directory to the one containing the schematic
- OBTAINED: xterm shell is opened at default home directory

StefanSchippers, 2023-11-08:
> ## Testing report
> ### Open symbol / schematic
>
> When switching from a symbol-less schematic (i.e. a testbench, which is never used as a sub-block), it
>
>     1. first shows an error message that the [...]

I think it is now fixed. I decided not to show any warnings.

> #### Issue: Menu item does not change
>
> Steps to reproduce:
>
>     * Perform steps of the above last issue
>
>     * In the empty symbol now shown, open the tab menu
>
>     * EXPECTED: menu item is called [...]

This is now fixed. Thank you for pointing this out!

> #### Issue: Warning dialog in case of already open counterparts
>
> Steps to reproduce:
>
>     * Open both sch and sym (i.e. using the tab menu or other means)
>
>     * The tab menu item is correct (in [...]

This should now work. if counterpard is already in a tab just switch to it like it does when clicking the counterpart tab.

> ### Open Terminal commands
>
> might have to do with my environment, but steps to reproduce:
>
>     * open schematic
>
>     * click "open circuit dir. term."
>
>     * EXPECTED: xterm shell is opened, and [...]

Unfortunately I can not reproduce this. Does this happen also for "Open sim. dir. term."? Are you using an old school terminal like xterm or rxvt or a "modern" one?
With Desktop Environment terminals i have had numerous issues in the past (just for the record I have attached a funny video of one of these problems) and I have blacklisted them all (well, I have blacklisted the DEs altogether), but if you want I can install for a moment one or more of these terminals and test.
So far I have tested with the lxterminal (was used in LXDE desktops, don't know if this environment still exists, I hope not) and things work as expected. 
Also verify if for some mistake or whatever reason there is no **duplicate schematic in the home (or current)  directory**.

https://github.com/StefanSchippers/xschem/assets/69359491/24956886-da12-4a49-829d-c3ff1d38c6ec

martinjankoehler, 2023-11-09:
> > ## Testing report
> > ### Open symbol / schematic
> > #### Issue: Error dialog with nonexistent counterparts
> 
> I think it is now fixed. I decided not to show any warnings.

Works now!
 
> > #### Issue: Menu item does not change
> This is now fixed. Thank you for pointing this out!

Works now!

> > #### Issue: Warning dialog in case of already open counterparts
> This should now work. if counterpart is already in a tab just switch to it like it does when clicking the counterpart tab.

Works now!

> > ### Open Terminal commands
> > might have to do with my environment, but steps to reproduce:
> Unfortunately I can not reproduce this. 
> Does this happen also for "Open sim. dir. term."? 
Yes, same for both.

> Are you using an old school terminal like xterm or rxvt or a "modern" one? 

xterm.

I'm pretty sure it has to do with our design flow scripts.
Which CLI commands are you launching for those menu items?

closed 2023-12-05 by StefanSchippers