StefanSchippers/xschem #134: after ctnr-f key-bindings stuck

olisnr, 2023-10-24:
if i select a few models via the `ctrl-f` search, and try to use `m` or `q` to move or change the selection, it wont work. but the menu-commands are working. 
is this some focus-problem? or is it intended?

StefanSchippers, 2023-10-24:
If you use ctrl-f the default is to just highlight the matching elements. You can change this by checking the Select checkbox. After selecting the matching elements I am able to move, delete or change ('q') their attributes. Can you please explain better what is your problem? in some cases the UI is blocked because there is some dialog box that has not been closed (may be obscured or minimized) , please check that too.

olisnr, 2023-10-24:
[image]

then i press `ctnr-f`

[image]

then i click `ok`

[image]

now all i wanted is selected, but if i press `q` nothing happens. but the menue `Properties/Edit` is working

olisnr, 2023-10-24:
ups, now i see: somehow the focus is changing to an other program... after closing the search box an other process has the focus. my be some cinnamon problem?

olisnr, 2023-10-24:
may be its because the dialog isnt modal?

StefanSchippers, 2023-10-24:
I will do some tests with other window managers.

StefanSchippers, 2023-10-24:
I tested with **fvwm3** (my window manager) and **icewm** (in its default configuration, which is click-to-focus) and xschem remains focused after a search (ctrl-f)

StefanSchippers, 2023-10-24:
Anyway I think the search dialog box does an unnecessary `grab set` (to force it to be in focus) that is probably unnecessary. This will make xschem responsive (to zoom/pan commands for example) even if the search dialog box is open. This dialog box is configured to remain on top of the xschem window, to avoid it to be obscured. Will remove this historic grab set, since I think it is not needed.

olisnr, 2023-10-24:
its switching to the system-monitor only if the `always on top` option is active. this seems to happen with every dialog-box, but it attract my attention only after the search.

StefanSchippers, 2023-10-24:
So if you close a dialog box in xschem (like the Load/Save As dialog) focus goes to another application? this is weird...

StefanSchippers, 2023-10-24:
Xschem Dialog boxes are configured to be on top of xschem window.
If xschem is set to 'always on top' may be there will be a contention of windows pretending to be on top, and the window manager detects the situation and moves focus away. Just a guess.

olisnr, 2023-10-24:
_So if you close a dialog box in xschem (like the Load/Save As dialog) focus goes to another application? this is weird..._

yes, but the focus goes only to an other application that has the `always on top` flag set.

olisnr, 2023-10-24:
if i also enable `always on top` for Xschem, then the dialogboxes will appear in the back of Xschem, but the focus still goes to the system-monitor after closing

StefanSchippers, 2023-10-24:
I see there are some issues on cinnamon with always on top windows. 
https://github.com/linuxmint/Cinnamon/issues/4386
https://github.com/linuxmint/Cinnamon/issues/4312
https://github.com/linuxmint/cinnamon/issues/10364
https://github.com/linuxmint/cinnamon/issues/10173

StefanSchippers, 2023-10-24:
I see there is (or was) a "`prevent focus stealing`" setting in the desktop. Is this available? can you test if it makes a difference?

olisnr, 2023-10-24:
there is no robbery-prevention anymore:

[image]

and "Attach dialog wi...". i will try it on the mate desktop...

olisnr, 2023-10-24:
now i installed `fvmw` and `mate` on linuX mint (ubuntu 22.04) and:
mate behaves like cinnamon
fvmw behaves correct

StefanSchippers, 2023-10-24:
Would be nice to see some other application which pops a dialog box and see if there is a similar problem. 
If not then I will like to investigate how to avoid that....

olisnr, 2023-10-24:
i think it could be some Tk/Tcl thing:
i downloaded an other program: `https://aplsimple.github.io/en/tcl/alited/index.html`...
and it does the same...

olisnr, 2023-10-24:
programs like `xed` dont do this

StefanSchippers, 2023-10-24:
Can you please test the following `test.tcl` program? it shows a main window and a button to open a dialog. Set the main window in focus (you will see a cursor in the text area) , then open the dialog box, close it and see if text area is still in focus.
Invoke the program with `wish test.tcl`
```
proc dialog {} {
  toplevel .a
  text .a.t
  .a.t insert 1.0 {Hello, World!}
  button .a.ok -text OK -command {destroy .a}
  pack .a.t .a.ok
  tkwait window .a
}

text .t
.t insert 1.0 {Main window}
button .close -text Close -command { destroy .}
button .dialog -text {Open dialog} -command {dialog}
pack .t
pack .dialog .close -side left

```

olisnr, 2023-10-25:
yes, this code also let the "always top" marked process steal the focus.

StefanSchippers, 2023-10-26:
I also tried setting some apps on top in ICEWM and i can't see this focus stealing problem.

https://github.com/StefanSchippers/xschem/assets/69359491/18a07f1b-0a8c-4d0d-83d5-a11e33b1dee0

StefanSchippers, 2023-10-26:
If you start xschem from a console, can you please issue this command:
xschem [~] **tk_focusFollowsMouse**
and see if the focus stealing still happens?

olisnr, 2023-10-26:
it dont change anything

https://github.com/StefanSchippers/xschem/assets/120017377/04255e7f-5caf-4161-a01d-aae22d9300fb

StefanSchippers, 2023-10-26:
Does the focus robbery happen with any application that is on the 'always on top' layer or only with the system monitor?
In the latter case are there options in System Monitor to change this?

StefanSchippers, 2023-10-26:
I think I need to do some tests on a live Cinnamon Mint image.

StefanSchippers, 2023-10-26:
OK, I got the same problem on Cinnamon, ( I also got the garbage on screen , but adding  the xorg.conf this was fixed right away)
I also found a fix for the problem. Please can you test this modified `test.tcl `program? run it as usual as `wish8.6 test.tcl`
The only change is the addition of `wm transient .a .` in the dialog creation routine. If that works I will apply this to all xschem dialogs.

```
proc dialog {} {
  toplevel .a -class dialog
  wm transient .a .
  text .a.t
  .a.t insert 1.0 {Hello, World!}
  button .a.ok -text OK -command {destroy .a}
  pack .a.t .a.ok
  update
  tkwait window .a
}

text .t
.t insert 1.0 {Main window}
button .close -text Close -command { destroy .}
button .dialog -text {Open dialog} -command {dialog}
pack .t
pack .dialog .close -side left
```

olisnr, 2023-10-26:
yes, this is a working Robbery Prevention...

olisnr, 2023-10-26:
the " I also got the garbage on screen" is the driver-problem? how did You get rid of it?

StefanSchippers, 2023-10-26:
For this machine (an asus core i7 with an intel GPU) it is a matter of disabling the PageFlip.
```
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
```

```
 Section "Device"
     Identifier  "intel"
     Driver      "intel"
     Option "PageFlip" "false"
 EndSection
```

StefanSchippers, 2023-10-26:
I will now start cinnamon on a samsung laptop with a AMD/radeon GPU and see how the patched xschem works.

StefanSchippers, 2023-10-26:
On the old Samsung I can set the xorg.con to work fine  either for xschem (no garbage in xschem, garbage in the desktop background) or for cinnamon (no garbage in the background, garbage in xschem)
The problem is that cinnamon uses OpenGL compositing by default and on these old computers OpenGL is full of bugs.
This computer is simply too old for a modern desktop Linux.

olisnr, 2023-10-27:
yes, i also think i would be good, if users dont need to be aware of the software-problems of there video-drivers. to the speed of Xschem: the biggest refresh break i saw until now are the graph-rectangles if data are loaded and plotted. 
i think the plots should be blanked while zooming or moving and redrawn only after the screen section.

closed 2023-11-02 by olisnr