StefanSchippers/xschem #135: Error message on exit when using no tabbed interface and replacekey

christoph-weiser, 2023-10-26:
xschem i showing an error upon exit when using a xschemrc that contains a combination
of the two statements.

```
# xschemrc
set tabbed_interface 0
set replace_key(Key-r) Control-e
```

The tool can exit via tcl console, but will not allow exit via window manager. 

xschem version: v3.4.3 
gnome-shell 43.6

[image]

StefanSchippers, 2023-10-26:
I think this is fixed upstream:

commit ae58077447be0649ae9ba8058ce8fd6d3283cd9f
Author: stefan schippers <stefan.schippers@gmail.com>
Date:   Thu Oct 12 21:09:45 2023 +0200

    move replace_key to global_array_list (it is an array var)

christoph-weiser, 2023-10-26:
Yes i can confirm that when I compile the latest commit on the master branch the issue is gone.

closed 2023-10-26 by christoph-weiser