StefanSchippers/xschem #136: question: is it possible to search models that are selected?

olisnr, 2023-10-26:
can i as example select all model=nmos that are currently selected or highlighted (or not selected) ? 

i think it this should at least be possible via the TCL command. but i dont found a documentation with the commands available.

StefanSchippers, 2023-10-26:
Depends what you are trying to achieve.
you can **select** all elements in a schematic by searching for `cell::type` and no value pattern.
You can then search and **unselect** all elements that have `cell::type` pattern set to `npn|pnp|resistor|capacitor`
this will for example remove all npn, pnp, resistors and capacitors from the selection.

you can write a small script to get all elements in current schematic:

```
foreach {name sym type}  [xschem instance_list] {
  puts "$name -- $sym -- $type"
}
```
you get this:
```p0 -- ipin.sym -- ipin
p3 -- ipin.sym -- ipin
xm2 -- nmos3.sym -- nmos
R7 -- res.sym -- resistor
xm1 -- nmos3.sym -- nmos
R0 -- res.sym -- resistor
...
...
```

you can iterate over the selected set and return the model associated to all selected instances (if they have one):

```
xschem [~] foreach i [xschem selected_set] { puts "$i --> [xschem getprop instance $i model]"}     
p0 --> 
p3 --> 
xm2 --> irf540
R7 --> 
xm1 --> irf540
R0 --> 
l8 --> 
R2 --> 
R3 --> 
p10 --> 
Q5 --> q2n2222
Q4 --> q2n2907p
R9 --> 
Q6 --> q2n2907p
R4 --> 
```

StefanSchippers, 2023-10-26:
SO I think you can iterate over the selection and build a list of elements with `model=nmos`, unselect everything (xschem unselect_all) and seleect elements in the list:
```
set mylist {}
foreach i [xschem selected_set] {
  set model [xschem getprop instance $i model]
 if {$model == {nmos}} {
    lappend mylist $i
  }
}
xschem unselect_all
foreach i $mylist {xschem select instance $i}
```

StefanSchippers, 2023-10-26:
may be try to evaluate this function:
```
# within a set of selected elements find and leave selected
# only elements with matching 'token=value'
# search can be exact or a regex expression, depending on 'search' parameter.
# search can be set to 'exact' (default) or 'regex'
# return number of matching selected items.
#
proc select_sub_selection {token value {search exact} } {
  set mylist {}
  foreach i [xschem selected_set] {
    set v [xschem getprop instance $i $token]
    if {$search == {exact}} {
      if {$v == $value} {
         lappend mylist $i
      }
    } else {
      if {[regexp $value $v]} {
         lappend mylist $i
      }
    }
  }
  xschem unselect_all
  foreach i $mylist {xschem select instance $i}
  return [xschem get lastsel]
}

```

olisnr, 2023-10-26:
yes! it works.

is there a documentation about the Xschems TCL commands, like `xschem selected_set` ?

StefanSchippers, 2023-10-26:
Yes, it is [here](https://xschem.sourceforge.io/stefan/xschem_man/developer_info.html#cmdref)
If you find something really unclear let me know.
Some commands are very low level, so if you don't understand the description (or there is almost no descrition) it is probably not needed.

closed 2023-10-26 by olisnr