StefanSchippers/xschem #137: CMOS example crashes if ...

olisnr, 2023-10-26:
... "Switch rawfile" is ctrl-clicked, before running the simulation. it says:

```
ich@P43s:/ALL/DoWN/xschem-master$ xschem
% rename dir (null) to /tmp/xschem_emergencysave_cmos_example_dfdddcbabe failed
EMERGENCY SAVE DIR: /tmp/xschem_emergencysave_cmos_example_dfdddcbabe

FATAL: signal 8
while editing: cmos_example
```

https://github.com/StefanSchippers/xschem/assets/120017377/c26518bb-2072-47e5-97a5-76c8a34e5be4

StefanSchippers, 2023-10-26:
Thank you, this is now fixed. The switch raw file works if there is at least one file loaded and there was no check for this.

StefanSchippers, 2023-10-26:
A note. Github tells me the mp4 file can not be played because it is corrupt. However I can download and see them.

olisnr, 2023-10-26:
does Github tell it or the web-browser? i see nothing. on chromium its working. 

thanks for the demo-fix!

closed 2023-10-26 by olisnr

StefanSchippers, 2023-10-26:
I see this on Firefox. Not sure if it is a browser or github issue. I think it is more a browser thing.
[image]

olisnr, 2023-11-04:
funny in my firefox version it also dont work... but in chromium it does