StefanSchippers/xschem #138: question about subcircuits

olisnr, 2023-10-28:
if i draw a schematics "opamap1.sch" and generate ans symbol "opamp1.sym", then the spice-netlist findes the subcircuit from the @symname. so its working if i have for every *.sch a *.sym. 
i wanted to use the same opamp_p.sym for differnent *.sch. the way i was trying is to change the global parameters:

```
type=subcircuit
format="@name @pinlist @xschem"
template="name=x1" @xschhem="opamp" 
```

and then set the properties in the schematics where i use the opamp_p.sym to `name=x2 xschem=opamap_p_130nm`. but this doesent work. because there is always the symbolname used to produce the .subckt in the spice netlist...

is there an other way to do this?

StefanSchippers, 2023-10-28:
Yes, there is an easy way to do that. You can do it even at instance level, so the same symbol placed multiple time can pick different schematics (schematics should have the same i/o interface of course, order of pins in schematic does not matter, the symbol ordering rules).
See this manual page, just the initial part.
https://xschem.sourceforge.io/stefan/xschem_man/tutorial_instance_based_implementation.html

olisnr, 2023-10-28:
cool!

closed 2023-10-28 by olisnr