StefanSchippers/xschem #139: Symbol lines dissapear when selected

Victor-Main, 2023-10-31:
I have observed this issue when working with Xschem. Whenever I select a component some of its lines will disappear after being selected . To my current observation this only happens when selecting lines that are not purely horizontal or vertical. For example, the circle present in voltage sources or the zig-zag lines on the resistor will disappear when selecting them. But a capacitor will remain untouched and the straight line that goes through the voltage source will also remain in place

For example with this simple low pass filter
[image]
I select the resistor
[image]
But upon unselecting it The resistor will disappear
[image]
And here is the same example but with a voltage source (the one above)
[image]

It is worth noting that updating the display will solve the issue. So moving the Xschem window, panning the view, or zooming in or out will once again redraw the component without issue.

I am using Xschem V3.4.5 and I have been able to observe this issue both in an ArchLinux and a Debian distro.

olisnr, 2023-10-31:
thats likley a GPU-driver problem. try the menue `option / Fix for GPUs with broken tiled fill`

Victor-Main, 2023-11-01:
Thank You. It did indeed fix the issue.

closed 2023-11-01 by Victor-Main

StefanSchippers, 2023-11-01:
I have now made the "fix" the default drawing method in xschem.