StefanSchippers/xschem #140: Installing ngspice _xschem and skywater130

Prabha1010, 2023-11-01:
I am getting this error which applying patch in Latest directory
[image]

Prabha1010, 2023-11-01:
And i get two different files listed one with.rej and another . acc

[image]

Prabha1010, 2023-11-01:
I also see that paths are not same in xschemrc file

[image]

Prabha1010, 2023-11-01:
And my xschem window has these layers which is entirely different from what your video says. Is it coz of earlier errors in installation?
[image]

StefanSchippers, 2023-11-01:
You followed a very old video. Please do the steps explained [here](https://xschem.sourceforge.io/stefan/xschem_man/tutorial_xschem_sky130.html) to install (in this order) magic, ngspice, open_pdks. For xschem follow the instructions [here](https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html).

Prabha1010, 2023-11-02:
Thanks Stefan.
But now when i followed all the steps, i get this 
[image]
[image]

StefanSchippers, 2023-11-02:
This means that you did not install xschem from the sources and you have an old ubuntu xschem installation.
As written in [this](https://xschem.sourceforge.io/stefan/xschem_man/tutorial_xschem_sky130.html) manual page:
**If you install xschem from sources ensure no xschem package is already installed in your linux system. Packaged xschem versions are too old so you should remove the installed package. The command for ubuntu/Debian systems is sudo apt-get remove --purge xschem** 

Then install xschem following [this](https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html) manual page.

closed 2023-12-11 by StefanSchippers